ARM-software/CMSIS_5 #727: Delays, waits, seconds vs ticks

tobermory, 2019-11-08:
In RTX4, any delay was to be specified in millis, e.g. osSignalWait() wanted timeout in millis. Unfortunately, a type somewhere in the call chain was U16 so the longest delay possible was 65535 millis, just 65 secs.

I am now moving to RTX5, and I see delays are now in ticks.  That means application code has to do the translation, e.g.

uint32_t tickInterval = 1000 / osKernelGetTickFreq();
uint32_t millisToWait = N;
osEventFlagsWait( .., N / tickInterval );

Surely every application level calculation of a time interval would use seconds/milliseconds? Datasheets mention 'after power up, allow 10ms for the widget to be ready'.  As I see it, with RTX5, I have to convert to ticks at every delay call site.

Related, sort of, the delay type in e.g. osEventFlagsWait is uint32_t, which suggests the upper limit of 65535 millis (or ticks as it is now) is no longer an issue.  The code tree looks a bit dense, but it does look like the low limit is solved in RTX5??

Stuart

JonatanAntoni, 2019-11-08:
Hi Stuart,

Short answer: Yes.

Long answer: I have to admit I do not understand your actual question :/

We have decided to go for native system ticks in CMSIS-RTOS2/RTX5 because that allows a more efficient implementation if the user can provide compile-time calculated timeout values. The downside is, as you already explained, you need to dynamically calculate the values on your own if you want your application to be agnostic of the actual system tick configuration.

Does this already address all your concerns?

Cheers,
Jonatan

ilg-ul, 2019-11-08:
> We have decided to go for native system ticks in CMSIS-RTOS2/RTX5 because that allows a more efficient implementation if the user can provide compile-time calculated timeout values

Fully agree.

tobermory, 2019-11-08:
Can you explain how I can achieve compile time delay values in ticks?  If my desired delay in millis is M, don't I have to compute M * osGetKernelTickFreq() / 1000 for every osXXX call accepting a delay parameter?  How can the compiler do that at build time?  I could just use M * OS_TICK_FREQ / 1000 but then I'm RTX-specific and not general-case RTOS2.

JonatanAntoni, 2019-11-11:
Hi @tobermory,

Yes, you are right. The very same compiler-time assumption holds true for other RTOS implementations as long as the tick frequency is known at compiler time. A very common tick frequency in M-class applications (and the RTX5 default) is 1kHz (1ms per tick).

If you are implementing a generic software component that shall be agnostic to the actual environment you need to do the calculation at rum-time.

Cheers,
Jonatan

tobermory, 2019-11-11:
OK, so to get compile time tick values I give us oskernelGetTickFreq() and use OS_TICK_FREQ instead, I can live with that.  

I don't actually use threads at all, I prefer to run one thread and have developed what is essentially a file-descriptor based, select+read model from POSIX/Unix.  I use event flags to implement select, so can wait on N I/O channels at once, with timeout.  All I then need is the idler thread doing a low power sleep. I find that the RTOS api, so holds  for RTX, don't actually solve the pressing problem, at least not in my (marine instrumentation) domain, which is one of a reactive system driven by I/O from many channels (GPS, Iridium modem, CTD, etc).  Multiple threads (one handling each device) mean critical regions and locks and introduce the nagging uncertainty of a deadlock. With a single thread, I can reason about the runtime behavior by looking at the source code, I'm not at the mercy of a thread scheduler. So I can run my OS_TICK_FREQ at 10ms, or even higher I think, depends on the resolution I need for delays and timing.  Most the I/O devices in my app work on the order of seconds. 

BTW I use the svc macros from rtx_core_cm.h to implement by open+close+read+write+select kernel calls, so am glad there are (in RTX5) in a header file, on RTX4 they weren't.

JonatanAntoni, 2019-11-12:
Sounds like an interesting application. Using multiple (prioritized) threads or not just depends on your requirements.

The timer thread should only get started if you are creating software timers. Otherwise it should not consume any resources. Implementing the tickless low power state is a bit tricky. There is an example in the docs and a more elaborated discussion in this forum. 

If you have any other question please consider raising separate issues for each topic. Does the above answer solve your current issue with delay?

Cheers,
Jonatan

closed 2020-03-26 by JonatanAntoni