ARM-software/CMSIS_5 #729: CMSIS 5.6.0 reports warnings in pc-lint

JuergenDuve, 2019-11-08:
Latest CMSIS 5.6.0 with special variables such as

#ifndef __INITIAL_SP
#define __INITIAL_SP Image$$ARM_LIB_STACK$$ZI$$Limit
#endif

cause warnings in pc-lint:

PC-lint for C/C++ (NT) Vers. 9.00k, Copyright Gimpel Software 1985-2013

--- Module: source\main.c (C)
#define __INITIAL_SP Image$$ARM_LIB_STACK$$ZI$$Limit
^
C:\Keil\ARM\CMSIS\Include\cmsis_armclang.h(124,0): Error 27: Illegal character (0x24)
^
C:\Keil\ARM\CMSIS\Include\cmsis_armclang.h(124,0): Error 27: Illegal character (0x24)

Please update the lint configuration, so that these armcc special $$ in variable names are ignored by pc-lint.

JonatanAntoni, 2019-11-08:
Hi Juergen,

thanks for addressing this.
Which PC-Lint configuration are you referring to?
I am not sure what we, in terms of CMSIS, can do about it. The `$` signs are used in linker generated symbols. From MISRA point of view those should of course not be used in regular code.

Cheers,
Jonatan

JuergenDuve, 2019-11-08:
Hi Jonatan,

My colleague Stefan dit a Siver issue and we got the following feedback:

cmsis 5.6.0 reports warnings in pc-lint CAS-140772-D5V3S7
Hello Stefan,

Thank you for finding this issue. I recommend going directly to the github page to report this issue. That will add visibility to this and help get it fixed by the CMSIS group faster.

https://github.com/ARM-software/CMSIS_5/issues

Best regards,

Kevin
about 21 hours ago
Modified on Nov-07-2019
Stefan Tauche

We are using the pclint config files as provided with MDK-ARM 5.27 under \UV4\Lint.

Best regards,
Jürgen

JonatanAntoni, 2019-11-08:
Hi Juergen,

I think that might have been a misunderstanding about the issue you tried to raise. From code perspective its CMSIS code that causes this PC-Lint warning, yes. But I think the code is perfectly right as the symbol using `$` signs is referring to linker generated symbols.

Ultimately, your question was to update the PC-Lint configuration so that these kind of symbols do not cause any warnings. The PC-Lint configuration you are probably referring to is shipped with MDK, not with CMSIS.

I have discussed with the MDK support colleagues and they'll address your request using the initial customer support case CAS-140772-D5V3S7. 

Cheers,
Jonatan

closed 2020-03-26 by JonatanAntoni