ARM-software/CMSIS_5 #743: startup_XXX.c generates -Werror=missing-attributes

fgr1986, 2019-11-19:
Compiling ` startup_ARMCM0plus.c` generates the following warning/error (if no -Werror): 

```
 error: 'NMI_Handler' specifies less restrictive attribute than its target 'Default_Handler': 'noreturn' [-Werror=missing-attributes]                                                                                                                                            51 | void NMI_Handler            (void) __attribute__ ((weak, alias("Default_Handler")));    ```

Tested with 
```                                                                                                                                                     GNU C17 (GNU Tools for Arm Embedded Processors 9-2019-q4-major) version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (arm-none-eabi)                                compiled by GNU C version 4.8.4, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP ```

JonatanAntoni, 2019-11-20:
Hi @fgr1986,

Thanks for pointing that out. Its a pitty, but I already suspected such feedback. 

The `Default_Handler` is implemented as an endless loop. Compilers tend to recognize that and warn about missing `noreturn` attribute. I guess there is no perfect solution.

I think we cannot add `noreturn` to the declaration of `NMI_Handler` because a user implementation might want to return. If we remove the `noreturn` from the `Default_Handler` compilers will warn about that.

Please note that the startup files are intended to be modified by the user, anyway. So you can simply remove all the default weak implementations and replace them with the actual implementations you need in your application.

Does that sound like a feasible solution for you?

Cheers,
Jonatan

fgr1986, 2019-11-20:
Hi @JonatanAntoni , thanks for your response.
It makes sense.

Thanks again!

closed 2019-11-21 by fgr1986