ARM-software/CMSIS_5 #744: Spurious signals at arm_sin_cos_q31 cosine output

ua1arn, 2019-11-21:
Then I use `arm_sin_cos_q31` in quadrature NCO - spectrum of cosine signal have additional frequencies, Sine output relative clear.
As illustration two pictures below. Shown NCO outputs at 700 Hz, fs=48 kHz.
[image]
[image]
If I use `arm_sin_q31` and` arm_sin_cos_q31` - no problems at all.

ua1arn, 2019-11-21:
Code for illustration:
```
typedef uint32_t ncoftw_t;
typedef int32_t ncoftwi_t;
// Convert ncoftw_t to q31 argument for arm_sin_cos_q31
// The Q31 input value is in the range [-1 0.999999] and is mapped to a degree value in the range [-180 179].
#define FTW2_SINCOS_Q31(angle) ((ncoftwi_t) (angle))
// Convert ncoftw_t to q31 argument for arm_sin_q31
// The Q31 input value is in the range [0 +0.9999] and is mapped to a radian value in the range [0 2*PI).
#define FTW2_COS_Q31(angle) ((q31_t) ((((ncoftw_t) angle) + 0x80000000uL) / 2))
#define IV ivqv [0]
#define QV ivqv [1]
typedef struct
{
	FLOAT_t ivqv [2];
} FLOAT32P_t;

static FLOAT32P_t getsincosf(ncoftw_t angle)
{
	FLOAT32P_t v;
	q31_t sincosv [2];
#if 1
	sincosv [0] = arm_sin_q31(FTW2_COS_Q31(angle));
	sincosv [1] = arm_cos_q31(FTW2_COS_Q31(angle));
#else
	arm_sin_cos_q31(FTW2_SINCOS_Q31(angle), & sincosv [0], & sincosv [1]);
	// at index 0 all fine
	// at index 1 with sidetones
#endif
	arm_q31_to_float(sincosv, v.ivqv, 2);
	return v;
}

```

christophe0606, 2019-11-21:
@ua1arn I was not able to reproduce a similar issue on my side and always get same results from arm_sin_cos_q31 and (arm_sin_q31 , arm_cos_q31).
I am not saying that there are no issues since obviously the implementations of arm_sin_cos_q31 and (arm_sin_q31 , arm_cos_q31) are quite different.

You C code looks right too.

Internally the functions are using the table sinTable_q31.
There is no corruption of this table ? 

Would it be possible for you to share the input pattern ?

ua1arn, 2019-11-21:
As I known, sinTable_q31 is a same for all functions?
angles generated by sequential calls of this code (not exact, unnesesary parts removed):

```
#define FTWROUND(ftw) ((uint32_t) (ftw))
FLOAT32P_t get_float_aflo(uint_fast8_t pathi)
{
	const ncoftw_t angle = angle_aflo [pathi];
	const FLOAT32P_t v = getsincosf(angle);
	angle_aflo [pathi] = FTWROUND(angle + anglestep_aflo [pathi]);
	return v;
}

```
Initial value of anglestep_aflo claculaed by
`#define FTWAF(freq) (((int_fast64_t) (freq) << NCOFTWBITS) / ARMI2SRATE)`
Where `NCOFTWBITS `= 32
`ARMI2SRATE `= 48000
`freq `may be 700 or (- 700)

christophe0606, 2019-11-22:
@ua1arn So, here are my latest experiments.
I have generated the angles using your code. The result looks correct (q31 converted to float).

[image]

From this, the reference spectrum I am computing is (we may use different normalization for the FFT).

[image]

When I apply your function FTW2_COS_Q31, I am getting the following angle. It looks correct.
[image]

For arm_sin_q31, I am getting this spectrum:

[image]

For arm_sin_cos_q31 (sin part), I am getting:
[image]

Both results look the same.

Nevertheless, there is an issue : the values are opposite. So something must be wrong in the angle computations.

[image]

I need to understand why the values are opposite as soon as I can find some time.

ua1arn, 2019-11-22:
> For arm_sin_cos_q31 (sin part), I am getting:
Spectrum problems appear only in **cosine part of result**.
Wrong angle not an issue for demonstration code, But thank you for pointing me to mistype.

christophe0606, 2019-11-22:
I've just rechecked with cos and I don't see any problem.

christophe0606, 2019-11-22:
When I compute the error between both signals (arm_cos_q31 and arm_sin_cos_q31 taking the cos), I get a SNR of -6 dB.

I have computed both spectrum with a BlackmannWindows and I indeed see that the spectrum of the arm_cos_q31 is less clean (so opposite of you since you see the problem with arm_sin_cos_q31).

When I switch to an HammingWindow I no more see (visually) any differences.

So I think the problem is not with the arm_sin_cos_q31 but with the window.

ua1arn, 2019-11-22:
I will reproduce these bugs with non-real time application and and then show results here.

christophe0606, 2019-12-17:
I close it for now. Feel free to reopen it if more data is available.

closed 2019-12-17 by christophe0606