ARM-software/CMSIS_5 #746: CMSIS-DAP : endpointaddress problem in example

windowsair, 2019-11-23:
Hi, I am trying to customize the CMSIS-DAP firmware with the path `CMSIS\DAP\Firmware\Examples\LPC-Link-II\V2\`, but the endpoint address makes me confused.

These endpoints are described in the keil site：
[http://www.keil.com/pack/doc/CMSIS/DAP/html/group__DAP__ConfigUSB__gr.html](url)
- Endpoint 1: Bulk Out – used for commands received from host PC.
- Endpoint 2: Bulk In – used for responses send to host PC.
- Endpoint 3: Bulk In (optional) – used for streaming SWO trace (if enabled with SWO_STREAM).

In fact, "endpoint 1" and "endpoint 2" work in the same endpoint(endpoint ), and "endpoint 3" works on endpoint 2.

[image]
The `Number` field in the image refers to the actual endpoint.

Then, according to the USB 2.0 specification:
```
Table 9-13. Standard Endpoint Descriptor

bEndpointAddress

Bit 3...0: The endpoint number
Bit 6...4: Reserved, reset to zero
Bit 7: Direction, ignored for
control endpoints
0 = OUT endpoint
1 = IN endpoint
```
Their endpoints should ***correspond to the values given below***:
```c
#define USBD_CUSTOM_CLASS0_IF0_EP1_BENDPOINTADDRESS 0x01
#define USBD_CUSTOM_CLASS0_IF0_EP2_BENDPOINTADDRESS 0x81
#define USBD_CUSTOM_CLASS0_IF0_EP3_BENDPOINTADDRESS 0x82
```
However, what is confusing is that `Firmware\Examples\LPC-Link-II\V2\RTE\USB\USBD_Config_CustomClass_0.h` ***is defined like this***:
```c
#define USBD_CUSTOM_CLASS0_IF0_EP1_BENDPOINTADDRESS    0x81
#define USBD_CUSTOM_CLASS0_IF0_EP2_BENDPOINTADDRESS    0x82
#define USBD_CUSTOM_CLASS0_IF0_EP3_BENDPOINTADDRESS    0x82
```

If you follow the above file, then the three endpoints are IN ENDPOINT, and the actual corresponding endpoint is also wrong.:disappointed:

I have no idea at all, I hope someone can point out the problem.:heart:

RobertRostohar, 2019-11-25:
The definitions `USBD_CUSTOM_CLASS0_IF0_EPx_BENDPOINTADDRESS` in USBD_Config_CustomClass_0.h use `x` as the internal index of the endpoint when constructing the USB descriptors. It starts with 0 and goes up to the max number of endpoints – 1. The index is not the physical endpoint address (the value of the definitions is).

In the example only the first three definitions are relevant:
```
#define USBD_CUSTOM_CLASS0_IF0_EP0_BENDPOINTADDRESS                0x01
#define USBD_CUSTOM_CLASS0_IF0_EP1_BENDPOINTADDRESS                0x81
#define USBD_CUSTOM_CLASS0_IF0_EP2_BENDPOINTADDRESS                0x82
```

Even easier should be to just use the Configuration Wizard and don’t bother with the actual defines. Take a look at the picture of the configuration in the [documentation](http://www.keil.com/pack/doc/CMSIS/DAP/html/group__DAP__ConfigUSB__gr.html#WinUSB).

windowsair, 2019-11-25:
Thank you for your answer. `USBD_CUSTOM_CLASS0_IF0_EP3_BENDPOINTADDRESS` is indeed an endpoint that is not being used. Since they are really confused with actual endpoints, I ignore this.

closed 2019-11-25 by windowsair