ARM-software/CMSIS_5 #756: I2C MasterReceive() - incomplete transfer event

sebastienmasson, 2019-12-03:
Dears,

Reading this page, https://www.keil.com/pack/doc/CMSIS/Driver/html/group__i2c__interface__gr.html#gafa22504bcf88a85584dfe6e0dd270ad5, I discover MasterReceive() is expected to generate the event ARM_I2C_EVENT_TRANSFER_INCOMPLETE.

How is this possible?  When can we expect such a situation happen?

Assuming the slave aborts the transfer, SDA will we released high; then, the master will keep going sending pulses on SCL and it will keep reading data.  Therefore, it will read 0xFF for each byte.
In this case, I do not see any mean for the master to detect the "incomplete transfer event".  Do you?

In case of bus error or in case of abitration lost, I can image the master to generate the appropriate event in addition to .._DONE and to ..._INCOMPLETE.

Would you please confirm the purpose of ..._INCOMPLETE ?

I thank you for your help
Seb

RobertRostohar, 2019-12-03:
The event ARM_I2C_EVENT_TRANSFER_INCOMPLETE should be generated when a transfer function finishes but not all the requested data is transferred.

It is not intended just for MasterReceive() but also for other transfer functions.

In case of MasterReceive() the .._INCOMPLETE event is set when address is not acknowledged from slave (.._ADDRESS_NAK), on bus error (.._BUS_ERROR) and in case of arbitration lost (.._ARBITRATION_LOST).
It does not provide any significant new information since other events are set in parallel. It could be used as a single check that not all data has been transferred without looking at all other possible events.

The .._ INCOMPLETE event is however more meaningful in case of MasterTransmit() where it is set also when slave does not acknowledge the data (with no other event except .._DONE).

sebastienmasson, 2019-12-03:
This confirms my assumption.  Thank you :-)

closed 2019-12-03 by sebastienmasson