ARM-software/CMSIS_5 #758: Will CMSIS DSP/NN lib support ARMv8.1-M Architecture?

tsaiyenting, 2019-12-04:
Hi,
Since Armv8.1-M Architecture might have significant performance uplift for machine learning and digital signal processing applications, will CMSIS DSP/NN library have AarmV8.1-M version?

Regards

christophe0606, 2019-12-05:
@tsaiyenting It is already the case and we are uploading new 8.1M implementations of the algorithms periodically.
It is of course a work in progress.

tsaiyenting, 2019-12-05:
Thanks for your reply.
Excuse me for asking: Is there an expected schedule?

christophe0606, 2019-12-05:
@tsaiyenting We do as quickly as we can. Today we have (for DSP) most of the functions. FilteringFunctions  are not yet finished.
TransformFunctions will come next.

tsaiyenting, 2019-12-05:
Thanks for your reply.
Look forward to your update :)

closed 2019-12-05 by tsaiyenting