ARM-software/CMSIS_5 #761: [Develop vs Master performance] arm_nn_read_q7x4_ia  slows down the execution compared to __SIMD32

antofara, 2019-12-05:
I recently swapped to the develop branch after I've been using the master branch for months. The first thing I noticed was a 5x increased execution time for my model containing many `arm_convolve_HWC_q7_fast_nonsquare`.

After some debug I reached the conclusion that the reason behind this slow-down is the replacement of expressions like `in = *__SIMD32(pIn)++;`  with `in = arm_nn_read_q7x4_ia(&pIn);`.

Indeed,if I replace back all the calls to arm_nn_read_q7x4_ia with __SIMD32 (like in master branch) in `arm_q7_to_q15_reordered_no_shift.c` and `arm_nnsupportfuctions.h:read_and_pad_reordered()`, the issue s fixed and the model is fast as before

Why has been done such a modification on the develop branch, isn't __SIMD32 safe?

Info about working environment:
Cortex M4 (Nordicsemi NRF52)
ARM GCC toolchain
O3 optimization

christophe0606, 2019-12-06:
@antofara It was done because of [issue 546](https://github.com/ARM-software/CMSIS_5/issues/546)

_SIMD32 was incorrect from C standard point of view and depending on the compiler it was sometimes leading to an incorrect behavior. It is indeed how the problem was detected (a few DSP kernels were giving incorrect results).

Replacing _SIMD32 by a more correct and portable sequence is of course impacting the performance. But the idea was that the compiler could optimize it out and the impact should be minimal.

It looks like it is not the case in your situation.

Can you tell me the gcc version you are using and if possible the compilation options in addition to the -O3 one ?

antofara, 2019-12-06:
@christophe0606 I tried with these two compilers:

arm-none-eabi-gcc.exe (GNU MCU Eclipse ARM Embedded GCC, 64-bit) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]

arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

here is an extract of the makefile regarding compilation flags and settings

```makefile
LIB_FILES += -larm_cortexM4lf_math
OPT = -O3 -g3
#OPT += -flto

CFLAGS += $(OPT)
CFLAGS += -DBSP_DEFINES_ONLY
CFLAGS += -DCONFIG_GPIO_AS_PINRESET

CFLAGS += -mcpu=cortex-m4
CFLAGS += -mthumb -mabi=aapcs
CFLAGS += -Wall 
CFLAGS += -ffunction-sections -fdata-sections -fno-strict-aliasing

CFLAGS += -fno-builtin -fshort-enums
CFLAGS += -include boards.h
CFLAGS += -include core_cm4.h
CFLAGS += -DARM_MATH_CM4
CFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16
CFLAGS += -DFLOAT_ABI_HARD

CXXFLAGS += $(OPT)

ASMFLAGS += -g3
ASMFLAGS += -mcpu=cortex-m4
ASMFLAGS += -mthumb -mabi=aapcs
ASMFLAGS += -DFLOAT_ABI_HARD
ASMFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16

# Linker flags
LDFLAGS += $(OPT)
LDFLAGS += -mthumb -mabi=aapcs -L$(SDK_ROOT)/modules/nrfx/mdk -L$(DSP_LIBS) -T$(LINKER_SCRIPT)
LDFLAGS += -mcpu=cortex-m4
LDFLAGS += -Wl,--gc-sections
LDFLAGS += --specs=nano.specs
LDFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16```

antofara, 2019-12-06:
@christophe0606 I found the following workaround:

```C
  __STATIC_FORCEINLINE void* arm_memcpy(void *dest, const void *src, size_t len)
 {
  char *d = dest;
  const char *s = src;
  while (len--)
    *d++ = *s++;
  return dest;
}
 
__STATIC_FORCEINLINE q31_t arm_nn_read_q7x4_ia(const q7_t **in_q7)
{
  q31_t val;
  arm_memcpy(&val, *in_q7, 4);
  *in_q7 += 4;

  return (val);
}
```

The compiler had difficulties in optimizing the call to th estandard memcpy routine. By using instead  `__STATIC_FORCEINLINE void* arm_memcpy` the performances are comparable with the one of the master branch

christophe0606, 2019-12-06:
@antofara Thanks for the workaround.
I have reported issue to compiler team. In the meantime, probably we are going to use a similar workaround to have better control on how the function is optimized.

sjoerdmeijer, 2019-12-06:
That looks fine as a short term workaround to me, but there is room for improving performance here as arm_memcpy can result in byte loads/stores (instead from just a word load/store). From a quick look, GCC does it fairly efficiently, but armclang is a bit worse. Also, I find it a lot of code just to load 4 bytes from memory. But briefly recapping, we do need resort to something like this because we can't do something like this:

`void foo (short *pd) {
    *p = (int *) pd;
    ..
  }`

I was hopeing that a read primitive like this:

`int read4(short **p) {
  int val;
  memcpy(&val, *p, 4);
  *p = *p + 2;
  return val;
}`

would just be lowered to:

`read4:
  ldr r2, [r0]
  mov r1, r0
  ldr r0, [r2], #4
  str r2, [r1]
  bx  lr`

That looks acceptable to me, and should be our goal. If I understood correctly, the problem is that GCC is not expanding the memcpy to just a word load and store. I haven't yet been able to reproduce this with a small example, but will continue a bit.

Maybe, this is a case where just a few intrinsics is fine to do just a load and store efficiently. I.e., perhaps it is find to have a generic memcpy function, but then also have to specialised read4 and read8 versions?

felix-johnny, 2020-04-06:
@antofara Have you tried without -fno-builtin ? There was a similar issue reported in TFLu https://github.com/tensorflow/tensorflow/issues/37361 .

felix-johnny, 2020-06-10:
If -fno-builtin doesn't work out please re-open the issue.

closed 2020-06-10 by felix-johnny