ARM-software/CMSIS_5 #762: About float matrix multiply efficiency

xiangxyq, 2019-12-06:
Hi everyone,
Platform :Arm M7 
Use MatrixFunctions/[arm_mat_mult_f32.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/MatrixFunctions/arm_mat_mult_f32.c), used this interface to compute neural network matrix multiply, found that the efficiency is less than below triple loop code:
`
for(int i = 0 ; i < A->mInfo.num_rows ; i++)
{
	for(int j = 0 ; j < B->mInfo.num_rows; j++)
	{
		result = 0.0f ;
		for(int k = 0 ; k < A->mInfo.num_cols ; ++k)  result += A1.data[i * A->mInfo.num_cols + k] * B1.data[ j* A->mInfo.num_cols + k] ;
		C.data[i * B->mInfo.num_rows + j] = result ;
	}
}
`
In the same condition(the same data), arm_mat_mult_f32.c cost about 457ms， but above code cost 300ms， the neural network dims almost (rows range:50~200) * (cols range: 20 ~ 200) 
Could anyone help me how to  optimize this issue?

Thanks!

hpycuit, 2019-12-06:
@JonatanAntoni I had the same problem.I've turned on the hardware float in the build configuration and used the GCC O3 optimization, but it's still pretty slow for a 200x200 large matrix.

christophe0606, 2019-12-17:
Which compilation options are you using ? Is ARM_MATH_LOOPUNROLL defined or not ?

closed 2021-09-08 by christophe0606