ARM-software/CMSIS_5 #766: Incorrect sample count for TEST_WEIGHTED_SUM_F32_3

stephanosio, 2019-12-13:
The `TEST_WEIGHTED_SUM_F32_3` test input pattern consists of only 10 elements, but `nbSamples` is set to 11.

https://github.com/ARM-software/CMSIS_5/blob/bb1d6260077283ef60e7b9d964557e2e9fbca972/CMSIS/DSP/Testing/Source/Tests/SupportTestsF32.cpp#L497-L505

https://github.com/ARM-software/CMSIS_5/blob/552cbefc2092ee7ed88f90f668a3b74b97bb0dea/CMSIS/DSP/Testing/desc.txt#L256

https://github.com/ARM-software/CMSIS_5/blob/ad00677da4fef6a3c85a2324882fa6bd82447184/CMSIS/DSP/Testing/Patterns/DSP/Support/SupportF32/Inputs6_f32.txt#L2

stephanosio, 2019-12-13:
cc @christophe0606

stephanosio, 2019-12-13:
```
starting test - test_arm_weighted_sum_f32_11
0x3ee6e5b3 0x3ed93754

    Assertion failed at ../src/f32.c:384: test_arm_weighted_sum_f32: (test_rel_error_f32(1, output, &ref[ref_offset], REL_ERROR_THRESH) is false)
incorrect computation result
FAIL - test_arm_weighted_sum_f32_11
```

christophe0606, 2019-12-16:
@stephanosio  Thanks for reporting this. It is indeed a bug in the test pattern generation script (pattern too short). I'll correct it.

christophe0606, 2019-12-16:
It should be corrected with commit 49c4ed79d274e9bfbc658c7adc7a71a4c2ca296d

closed 2019-12-17 by christophe0606