ARM-software/CMSIS_5 #770: [I2C] MasterTransmit() and MasterReceive() behaviour with xfer_pending=true

sebastienmasson, 2019-12-16:
Dears,

Would you please help to understand the documentation when xfer_pending=true?

The documentation says "Set the parameter xfer_pending to true is another transfer operation follows" [...] "When transmit/receive opereation has finished the ARM_..._DONE event is generated".

How should I understand "the operation has finished"?  Is the operation finished when all data bytes have been transfered?  or, when a stop condition occured (= when the last operation has finished)?
Do you expect an event to be generated between every transfer operations?

I thank you for your help,
BR
Seb

RobertRostohar, 2019-12-17:
xfer_pending only controls if STOP is generated. Otherwise the operation is the same.

From the documentation:

> The operation consists of:
> - Master generates START condition
> - Master addresses the Slave as Master Transmitter
> - Master transmits data to the addressed Slave
> - Master generates STOP condition (if xfer_pending is "false")

ARM_..._DONE event is generated either after STOP (xfer_pending = false) or after requested data has been transmitted.

sebastienmasson, 2019-12-17:
Hi Robert,

I understand the purpose of xfer_pending, my question relates to the ..._DONE event.  I wish to be sure I understand it the right way to ensure the driver I am writing respects the specifications.

Are you saying the following?
* if xfer_pending = false; then, ..._DONE is generated after the stop condition;
* If xfer_pending = true; then, ..._DONE is generated after the data have been sent/received.
Which means, a ..._DONE event could be generated each time the transfer direction change, for instance; or, typically, twice when the 7-bit combined format is used)

Or,

Are you saying the following?
* if xfer_pending = false; then, ..._DONE is generated after the stop condition;
* if xfer_pending = true; then, I can choose to generate ..._DONE
   - each time the n requested bytes have been sent/received; or,
   - only once when the stop condition finally occured.

There might have a dedicated documentation for drivers developers.  Is there?  I am struggling to extract software specifications from the end-users' documentation.  This would really help.

I really appreciate your help :-)

BR,
Seb

RobertRostohar, 2019-12-17:
The first one. The …_DONE event corresponds to a single operation triggered by MasterTransmit.

Let's take a look at an example for random read of I2C EEPROM (24 series):

```
int32_t EEPROM_ReadBuf (uint16_t addr, uint8_t *buf, uint32_t len) {
  uint8_t a[2];

  a[0] = (uint8_t)(addr >> 8);
  a[1] = (uint8_t)(addr & 0xFF);

  ptrI2C->MasterTransmit (DeviceAddr, a, 2, true);
  while (ptrI2C->GetStatus().busy);
  ptrI2C->MasterReceive (DeviceAddr, buf, len, false);
  while (ptrI2C->GetStatus().busy);
  if ((uint32_t)ptrI2C->GetDataCount () != len) return -1;

  return 0;
}
```

Completion of MasterTransimt is checked with busy flag in a pooling loop. This mechanism could be replaced with waiting for …_DONE event. For each operation (MasterTransmit/Receive) a single ..._DONE event should be generated.

sebastienmasson, 2019-12-19:
Pretty clear.  Thank you :-)

closed 2019-12-19 by sebastienmasson