ARM-software/CMSIS_5 #773: Uninitialised variable warning in arm_spline_interp_f32.c

stephanosio, 2019-12-17:
An uninitialised variable warning is displayed when compiling `arm_spline_interp_f32.c`.

```
[150/192] Building C object zephyr/ext/hal/cmsis/DSP/Sourc...__Source__SupportFunctions.dir/arm_spline_interp_f32.c.ob /home/stephanos/Dev/zephyr-public/zephyr/ext/hal/cmsis/DSP/Source/SupportFunctions/arm_spline_interp_f32.c: In function 'arm_spline_f32':
/home/stephanos/Dev/zephyr-public/zephyr/ext/hal/cmsis/DSP/Source/SupportFunctions/arm_spline_interp_f32.c:321:76: warning: 'di' may be used uninitialized in this function [-Wmaybe-uninitialized]
  321 |         *pDst = y[i-1]+bi*(x_sc-x[i-1])+c[i]*(x_sc-x[i-1])*(x_sc-x[i-1])+di*(x_sc-x[i-1])*(x_sc-x[i-1])*(x_sc-x[i-1]);
      |                                                                          ~~^~~~~~~~~~~~~~
/home/stephanos/Dev/zephyr-public/zephyr/ext/hal/cmsis/DSP/Source/SupportFunctions/arm_spline_interp_f32.c:321:26: warning: 'bi' may be used uninitialized in this function [-Wmaybe-uninitialized]
  321 |         *pDst = y[i-1]+bi*(x_sc-x[i-1])+c[i]*(x_sc-x[i-1])*(x_sc-x[i-1])+di*(x_sc-x[i-1])*(x_sc-x[i-1])*(x_sc-x[i-1]);
      |                        ~~^~~~~~~~~~~~~~
```

stephanosio, 2019-12-17:
cc @christophe0606

christophe0606, 2019-12-17:
@stephanosio Which compiler ? gcc ? AC6 ?
I am currently checking build issues with different compilers and different build config.
So, I'll correct this as part of this work.

stephanosio, 2019-12-17:
@christophe0606 I am using GCC 9.2.0.

This warning is generated because the `di` and `bi` variables are uninitialised when `n <= 1`:

https://github.com/ARM-software/CMSIS_5/blob/49c4ed79d274e9bfbc658c7adc7a71a4c2ca296d/CMSIS/DSP/Source/SupportFunctions/arm_spline_interp_f32.c#L243-L247

It might be a good idea to add an assertion like the following:
```
ASSERT(n > 1);
```

christophe0606, 2019-12-17:
@stephanosio I have made a temporary fix. I have initialized those variables to remove the warning. I have added a comment in the init function to highlight the fact that n must be > 1

And in a next release, this function will do the required checks (there is more than just checking n > 1) and return an error message if the pre-conditions are not true.

closed 2019-12-18 by christophe0606