ARM-software/CMSIS_5 #774: arm_merge_sort_core_f32 writes to the input data buffer

stephanosio, 2019-12-18:
`arm_merge_sort_f32` (`arm_merge_sort_core_f32`) writes to the input data buffer (`pSrc`) when performing a merge sort.

### Console Output
```
starting test - test_arm_sort_out_merge_11
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x408324
E: r0/a1:  0x00000000  r1/a2:  0xbe496d11  r2/a3:  0x00000005
E: r3/a4:  0xbf5fe006 r12/ip:  0x7fdffa26 r14/lr:  0x004010f3
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x0040751e
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20400048 (unknown)
E: Halting system
```

### Faulting Instruction Trace
```
>>> disas /m 0x0040751e
Dump of assembler code for function arm_merge_sort_core_f32:
33      static void topDownMerge(float32_t * pA, uint32_t begin, uint32_t middle, uint32_t end, float32_t * pB, uint8_t dir)

34      {
35          /* Left  array is pA[begin:middle-1]
36           * Right Array is pA[middle:end-1]
37           * They are merged in pB
38           */
39
40          uint32_t i = begin;

41          uint32_t j = middle;

42          uint32_t k;

43
44          // Read all the elements in the sublist
45          for (k = begin; k < end; k++)

46          {
47              // Merge
48              if (i < middle && (j >= end || dir==(pA[i] <= pA[j])) )
   0x004074e8 <+72>:    cmp     r7, r4
   0x004074ea <+74>:    bls.n   0x407516 <arm_merge_sort_core_f32+118>
   0x004074ec <+76>:    cmp     r5, r10
   0x004074ee <+78>:    ldr.w   r1, [r6, r4, lsl #2]
   0x004074f2 <+82>:    bls.n   0x40750a <arm_merge_sort_core_f32+106>
   0x004074f4 <+84>:    ldr.w   r0, [r6, r10, lsl #2]
   0x004074f8 <+88>:    str     r1, [sp, #12]
   0x004074fa <+90>:    bl      0x401138 <__aeabi_fcmpge>
   0x004074fe <+94>:    subs    r0, #0
   0x00407500 <+96>:    ldr     r1, [sp, #12]
   0x00407502 <+98>:    it      ne
   0x00407504 <+100>:   movne   r0, #1
   0x00407506 <+102>:   cmp     r0, r11
   0x00407508 <+104>:   bne.n   0x407516 <arm_merge_sort_core_f32+118>

49              {
50                  pB[k] = pA[i];
   0x0040750a <+106>:   adds    r4, #1
   0x0040750c <+108>:   str.w   r1, [r8, r9, lsl #2]

51                  i++;
   0x00407510 <+112>:   add.w   r9, r9, #1
   0x00407514 <+116>:   b.n     0x4074de <arm_merge_sort_core_f32+62>

52              }
53              else
54              {
55                  pB[k] = pA[j];
   0x00407516 <+118>:   ldr.w   r3, [r6, r10, lsl #2]
   0x0040751a <+122>:   add.w   r10, r10, #1
   0x0040751e <+126>:   str.w   r3, [r8, r9, lsl #2]

56                  j++;
   0x00407522 <+130>:   b.n     0x407510 <arm_merge_sort_core_f32+112>

57              }
58          }
59      }
60
61      static void arm_merge_sort_core_f32(float32_t * pB, uint32_t begin, uint32_t end, float32_t * pA, uint8_t dir)
62      {

63          if((int32_t)end - (int32_t)begin >= 2 )           // If run size != 1 divide
   0x004074a0 <+0>:     stmdb   sp!, {r4, r5, r6, r7, r8, r9, r10, r11, lr}
   0x004074a4 <+4>:     mov     r8, r3
   0x004074a6 <+6>:     subs    r3, r2, r1
   0x004074a8 <+8>:     sub     sp, #20
   0x004074aa <+10>:    mov     r6, r0
   0x004074ac <+12>:    cmp     r3, #1
   0x004074ae <+14>:    mov     r4, r1
   0x004074b0 <+16>:    mov     r5, r2
   0x004074b2 <+18>:    ldrb.w  r11, [sp, #56]  ; 0x38
   0x004074b6 <+22>:    ble.n   0x4074e2 <arm_merge_sort_core_f32+66>

64          {
65              int32_t middle = (end + begin) / 2;           // Take the middle point
   0x004074b8 <+24>:    adds    r7, r2, r1
   0x004074ba <+26>:    mov     r3, r0
   0x004074bc <+28>:    str.w   r11, [sp]
   0x004074c0 <+32>:    mov     r0, r8
   0x004074c2 <+34>:    lsrs    r7, r7, #1

66
67              arm_merge_sort_core_f32(pA, begin,  middle, pB, dir);  // Sort the left part
   0x004074c4 <+36>:    mov     r9, r4
   0x004074c6 <+38>:    mov     r2, r7
   0x004074c8 <+40>:    mov     r10, r7
   0x004074ca <+42>:    bl      0x4074a0 <arm_merge_sort_core_f32>

68              arm_merge_sort_core_f32(pA, middle,    end, pB, dir);  // Sort the right part
   0x004074ce <+46>:    mov     r3, r6
   0x004074d0 <+48>:    mov     r2, r5
   0x004074d2 <+50>:    mov     r1, r7
   0x004074d4 <+52>:    mov     r0, r8
   0x004074d6 <+54>:    str.w   r11, [sp]
   0x004074da <+58>:    bl      0x4074a0 <arm_merge_sort_core_f32>

69
70              topDownMerge(pB, begin, middle, end, pA, dir);
   0x004074de <+62>:    cmp     r5, r9
   0x004074e0 <+64>:    bhi.n   0x4074e8 <arm_merge_sort_core_f32+72>
   0x004074e2 <+66>:    add     sp, #20
   0x004074e4 <+68>:    ldmia.w sp!, {r4, r5, r6, r7, r8, r9, r10, r11, pc}

End of assembler dump.
```

### Faulting Memory Address Trace
```
0x408324 <in_sort+12>:  0xbe496d11
```

### Memory Map
```
 .rodata.in_sort
                0x0000000000408318       0x2c app/libapp.a(f32.c.obj)
```

stephanosio, 2019-12-18:
cc @@clamar01 @christophe0606

stephanosio, 2019-12-18:
I believe a separate temporary/scratch buffer should be specified for this, as was done in `arm_logsumexp_dot_prod_f32`.

_p.s. all the other sort functions seem to treat `pSrc` as `const float32_t *` and I suppose that is indeed the intended behaviour?_

christophe0606, 2020-01-07:
It should be solved by latest @clamar01 commit 3bd171476dcafbd9ebdab8acf3b04456a35dbe4c

closed 2020-01-17 by christophe0606