ARM-software/CMSIS_5 #786: Under-sized state buffer for FIRQ15 test

stephanosio, 2019-12-27:
The state buffer for the `FIRQ15` test is under-sized.

https://github.com/ARM-software/CMSIS_5/blob/73ed0410119f500ec5025c79c04f46379c46620e/CMSIS/DSP/Testing/Source/Tests/FIRQ15.cpp#L126

stephanosio, 2019-12-27:
cc @christophe0606

stephanosio, 2019-12-27:
In addition, the following comment is not correct because the minimum state buffer size when `ARM_MATH_DSP` is enabled is `blockSize + numTaps`:

https://github.com/ARM-software/CMSIS_5/blob/73ed0410119f500ec5025c79c04f46379c46620e/CMSIS/DSP/Testing/Source/Tests/FIRQ15.cpp#L123-L125

stephanosio, 2019-12-27:
The same problem exists in `FIRQ7`, `FIRQ31` and `FIRF32` tests as well.

christophe0606, 2020-01-07:
It should be solved with commit 1d1d70e2385b7c8c31292669713270b7bb378052

closed 2020-01-17 by christophe0606