ARM-software/CMSIS_5 #789: SNR below threshold for TransformQ15 test

stephanosio, 2020-01-01:
## Summary

The SNR value reported by the TransformQ15 IFFT tests are below the specified threshold (i.e. 30).

https://github.com/ARM-software/CMSIS_5/blob/73ed0410119f500ec5025c79c04f46379c46620e/CMSIS/DSP/Testing/Source/Tests/TransformQ15.cpp#L8

## Notes

- There is no problem with the (forward) FFT tests; only the IFFT tests report abnormally low SNR.
- All the other tests, including TransformQ31, report reasonable SNR and satisfy the `SNR_THRESHOLD` specified for each test.

## Log

_Ignore test `PASS` status, as the SNR assertion was deliberately disabled_

**QEMU Cortex-M33 Target**
```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m33
qemu-system-arm: warning: nic lan9118.0 has no peer
*** Booting Zephyr OS build zephyr-v2.1.0-594-g17bab9c7b999  ***
Running test suite transform_q15
===================================================================
starting test - test_arm_cfft_q15_noisy_16
SNR = 68.607925
PASS - test_arm_cfft_q15_noisy_16
===================================================================
starting test - test_arm_cifft_q15_noisy_16
SNR = 56.335636
PASS - test_arm_cifft_q15_noisy_16
===================================================================
starting test - test_arm_cfft_q15_noisy_32
SNR = 59.494591
PASS - test_arm_cfft_q15_noisy_32
===================================================================
starting test - test_arm_cifft_q15_noisy_32
SNR = 44.092136
PASS - test_arm_cifft_q15_noisy_32
===================================================================
starting test - test_arm_cfft_q15_noisy_64
SNR = 61.709412
PASS - test_arm_cfft_q15_noisy_64
===================================================================
starting test - test_arm_cifft_q15_noisy_64
SNR = 45.335495
PASS - test_arm_cifft_q15_noisy_64
===================================================================
starting test - test_arm_cfft_q15_noisy_128
SNR = 52.768990
PASS - test_arm_cfft_q15_noisy_128
===================================================================
starting test - test_arm_cifft_q15_noisy_128
SNR = 32.719532
PASS - test_arm_cifft_q15_noisy_128
===================================================================
starting test - test_arm_cfft_q15_noisy_256
SNR = 55.635139
PASS - test_arm_cfft_q15_noisy_256
===================================================================
starting test - test_arm_cifft_q15_noisy_256
SNR = 32.935146
PASS - test_arm_cifft_q15_noisy_256
===================================================================
starting test - test_arm_cfft_q15_noisy_512
SNR = 46.698872
PASS - test_arm_cfft_q15_noisy_512
===================================================================
starting test - test_arm_cifft_q15_noisy_512
SNR = 20.629599
PASS - test_arm_cifft_q15_noisy_512
===================================================================
[... 230 lines cut ...]
===================================================================
starting test - test_arm_cifft_q15_step_1024
SNR = 25.809500
PASS - test_arm_cifft_q15_step_1024
===================================================================
starting test - test_arm_cfft_q15_step_2048
SNR = 47.052162
PASS - test_arm_cfft_q15_step_2048
===================================================================
starting test - test_arm_cifft_q15_step_2048
SNR = 13.337952
PASS - test_arm_cifft_q15_step_2048
===================================================================
starting test - test_arm_cfft_q15_step_4096
SNR = 50.012596
PASS - test_arm_cfft_q15_step_4096
===================================================================
starting test - test_arm_cifft_q15_step_4096
SNR = 13.383279
PASS - test_arm_cifft_q15_step_4096
===================================================================
Test suite transform_q15 succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

christophe0606, 2020-01-07:
In the file Testing/desc.txt describing the tests, some tests are disabled:
```
disabled {cifft_noisy_512_q15:test_cifft_q15}
disabled {cifft_noisy_1024_q15:test_cifft_q15}
disabled {cifft_noisy_2048_q15:test_cifft_q15}
disabled {cifft_noisy_4096_q15:test_cifft_q15}
disabled {cifft_step_512_q15:test_cifft_q15}
disabled {cifft_step_1024_q15:test_cifft_q15}
disabled {cifft_step_2048_q15:test_cifft_q15}
disabled {cifft_step_4096_q15:test_cifft_q15}
```

I think this is corresponding to the failing tests you report above. So, it is normal.

The Q15 ifft is not accurate enough and will fail on big input patterns.
Until the implementation is reworked, those tests will fail and thus are disabled.

stephanosio, 2020-01-07:
> In the file Testing/desc.txt describing the tests, some tests are disabled:

Ok, that makes a lot of sense. Thanks for clarifying.

closed 2020-01-07 by stephanosio