ARM-software/CMSIS_5 #790: LLVM clang support

4a-42, 2020-01-02:
I have a cmake-based project targeting arm cortex-m4, and I want to run clang-tidy on my code. 

When I attempt to run clang-tidy, I get this message
```
CMSIS_5/CMSIS/Core/Include/cmsis_compiler.h:278:4: error: Unknown compiler. [clang-diagnostic-error]
  #error Unknown compiler.
```

my version of clang has `__clang__` instead of `__ARMCC_VERSION`

maybe there's an option 
`set(CMAKE_CXX_CLANG_TIDY clang-tidy --header-filter=* -checks=*)`
I could set to get this to work for the purposes of static analysis? clang-tidy doesn't need to link/execute, it just needs to be able to compile the code.

As a side note, I think a good feature request could be to formally add support for llvm clang, since it appears to be an unsupported compiler.

4a-42, 2020-01-06:
for the cmake option to get clang-tidy to work, the workaround is 
```cmake
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
set(CMAKE_C_CLANG_TIDY clang-tidy --header-filter=* -checks=* --extra-arg=-U_WIN32 --extra-arg=-D__GNUC__)  
set(CMAKE_CXX_CLANG_TIDY clang-tidy --header-filter=* -checks=* --extra-arg=-U_WIN32 --extra-arg=-D__GNUC__)
```
`-U_WIN32` doesn't affect cmsis, but it does affect my mdk.

JonatanAntoni, 2020-01-09:
Hi @4a-42,

we have only upstream GCC support in CMSIS so far, no upstream clang.
If you would like to see upstream clang support in CMSIS please could you help us getting it in place? Probably we need to add another compiler header, similar to `cmsis_armclang.h` (e.g. `cmsis_clang.h`). I am happy to accept a pull-request adding this.

Cheers,
Jonatan

4a-42, 2020-06-19:
I decided to look back into this for a while today. I'm posting this as a way to have this written down so I can continue looking into this. 

Fortunately clang and gcc are largely compatible, so it shouldn't be too much work to copy paste `cmsis_gcc.h` into a new file for llvm clang. The only incompatibility I've had so far from messing around had to do with the `__set_FPSCR` function. My workaround looked like this, but with a proper compiler header, it wouldn't be necessary:

```c
__attribute__( ( always_inline ) ) __STATIC_INLINE void __set_FPSCR(uint32_t fpscr)
{
#if (__FPU_PRESENT == 1U) && (__FPU_USED == 1U)
  /* Empty asm statement works as a scheduling barrier */
  __ASM volatile ("");
  #ifndef __LINT__
  __ASM volatile ("VMSR fpscr, %0" : : "r" (fpscr) : "vfpcc");
  #endif
  __ASM volatile ("");
#endif
}
``` 

My use case is to get clang-tidy to work but it would be cool to run the library's tests on the clang compiler to see if it can be used to compile firmware. 

Anyways, I'll go ahead and write down my toolchain info here:

```bash
/mnt/c/Program Files/LLVM/bin> ./clang.exe --version
clang version 10.0.0
Target: x86_64-pc-windows-msvc
Thread model: posix
InstalledDir: C:\Program Files\LLVM\bin
```

based off of this stackoverflow post: <https://stackoverflow.com/questions/2224334/gcc-dump-preprocessor-defines>

clang C preprocessor defines (targeting an arm cortex-m chip):

```bash
/mnt/c/Program Files/LLVM/bin> ./clang.exe --target=armv7m-none-eabi -dM -E - < /dev/null
#define _ILP32 1
#define __APCS_32__ 1
#define __ARMEL__ 1
#define __ARM_32BIT_STATE 1
#define __ARM_ACLE 200
#define __ARM_ARCH 7
#define __ARM_ARCH_7M__ 1
#define __ARM_ARCH_EXT_IDIV__ 1
#define __ARM_ARCH_ISA_THUMB 2
#define __ARM_ARCH_PROFILE 'M'
#define __ARM_EABI__ 1
#define __ARM_FEATURE_CLZ 1
#define __ARM_FEATURE_IDIV 1
#define __ARM_FEATURE_LDREX 0x7
#define __ARM_FEATURE_QBIT 1
#define __ARM_FEATURE_SAT 1
#define __ARM_FP16_ARGS 1
#define __ARM_FP16_FORMAT_IEEE 1
#define __ARM_PCS 1
#define __ARM_SIZEOF_MINIMAL_ENUM 4
#define __ARM_SIZEOF_WCHAR_T 4
#define __ATOMIC_ACQUIRE 2
#define __ATOMIC_ACQ_REL 4
#define __ATOMIC_CONSUME 1
#define __ATOMIC_RELAXED 0
#define __ATOMIC_RELEASE 3
#define __ATOMIC_SEQ_CST 5
#define __BIGGEST_ALIGNMENT__ 8
#define __BYTE_ORDER__ __ORDER_LITTLE_ENDIAN__
#define __CHAR16_TYPE__ unsigned short
#define __CHAR32_TYPE__ unsigned int
#define __CHAR_BIT__ 8
#define __CHAR_UNSIGNED__ 1
#define __CLANG_ATOMIC_BOOL_LOCK_FREE 2
[... 708 lines cut ...]
#define __llvm__ 1
#define __private_extern__ extern
#define __thumb2__ 1
#define __thumb__ 1
```

I'm worried whether armclang also defines `__clang__` and/or `__llvm__`. 

Unfortunately, I don't have an armclang license, but I found this in `armclang Reference > armclang Command-line Options > -E`:

```txt
1.4 -E
Executes the preprocessor step only.
By default, output from the preprocessor is sent to the standard output stream and can be redirected to a
file using standard UNIX and MS-DOS notation.
You can also use the -o option to specify a file for the preprocessed output.
By default, comments are stripped from the output. Use the -C option to keep comments in the
preprocessed output.
Examples
Use -E -dD to generate interleaved macro definitions and preprocessor output:
armclang --target=aarch64-arm-none-eabi -mcpu=cortex-a53 -E -dD source.c > raw.c
Use -E -dM to list all the macros that are defined at the end of the translation unit, including the
predefined macros:
armclang --target=arm-arm-none-eabi -mcpu=cortex-m3 -E -dM source.c
```

JonatanAntoni, 2020-06-19:
Hi @4a-42,

thanks for taking this one. Here are the outputs from Arm Compiler 6.14:

`armclang.exe --target=arm-arm-none-eabi -mcpu=cortex-m3  -dM -E - < /dev/null`
[armclang-c.txt](https://github.com/ARM-software/CMSIS_5/files/4803906/armclang-c.txt)

`armclang.exe --target=arm-arm-none-eabi -mcpu=cortex-m3  -dM -E -x c++ - < /dev/null` 
[armclang-cpp.txt](https://github.com/ARM-software/CMSIS_5/files/4803909/armclang-cpp.txt)

Cheers,
Jonatan

JonatanAntoni, 2021-06-08:
Hi @4a-42,

Are you still keen on getting upstream clang support into CMSIS? Do you volunteer to provide the required `cmsis_clang.h` header (may be based on existing cmsis_armclang.h` file? Or shall we rather close this?

Thanks,
Jonatan

flit, 2021-06-08:
I'd like to see support for upstream clang in CMSIS-Core. 👍🏽 

As @4a-42 describes, the `cmsis_clang.h` header would be basically a copy of `cmsis_gcc.h` rather than based on `cmsis_armclang.h`. armclang has a lot of instruction intrinsics defined as `__builtin_arm_foo()`, though oddly not all instructions have such intrinsics. The main difference between gcc and clang headers would be the pragmas. Program startup, stack/heap symbols, etc would follow the gcc style rather than armclang.

This brings up the question of whether (most of) the intrinsic definitions within `cmsis_gcc.h` can be shared for upstream clang. Duplicating 90% of the file contents seems like a bad idea.

JonatanAntoni, 2021-06-09:
Thanks, @flit. Reusing `cmsis_gcc.h` for clang may be a bit confusing as well. I don't have a clear solution, yet. I am happy to receive contribution without any strong opinion how to structure the files.

flit, 2021-06-09:
Well, we can just start with duplicate intrinsics, then improve upon that if we come up with a clean solution.

escherstair, 2023-11-07:
Hi,
is this activity in progress or is it stale?

One more thing:
> for the cmake option to get clang-tidy to work, the workaround is
>
> ```cmake
> set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
> set(CMAKE_C_CLANG_TIDY clang-tidy --header-filter=* -checks=* --extra-arg=-U_WIN32 [...]

since Arm Compiler 6 is clang-based, is it possible using the following workaround?
```cmake
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
set(CMAKE_C_CLANG_TIDY clang-tidy --header-filter=* -checks=* --extra-arg=-U_WIN32 --extra-arg=-D__ARMCC_VERSION --extra-arg=-D__ARMCC_VERSION=6100100)  
set(CMAKE_CXX_CLANG_TIDY clang-tidy --header-filter=* -checks=* --extra-arg=-U_WIN32 --extra-arg=-D__ARMCC_VERSION --extra-arg=-D__ARMCC_VERSION=6100100)
```

JonatanAntoni, 2023-11-07:
We are working on next major release of CMSIS here: https://github.com/ARM-software/CMSIS_6
It contains preliminary Clang support. Feel free to give it a try and let us know your feedback.

4a-42, 2023-11-07:
I remember that setting `-fgnuc-version=<arg>` to a more modern version helped with some things in the headers.

escherstair, 2023-11-07:
> We are working on next major release of CMSIS here: https://github.com/ARM-software/CMSIS_6 It contains preliminary Clang support. Feel free to give it a try and let us know your feedback.

I try to do this.
Can I install CMSIS_6 using [CMSIS-Toolbox](https://github.com/Open-CMSIS-Pack/cmsis-toolbox) or should I clone the github repo?

JonatanAntoni, 2023-11-07:
> I try to do this. Can I install CMSIS_6 using [CMSIS-Toolbox](https://github.com/Open-CMSIS-Pack/cmsis-toolbox) or should I clone the github repo?

There is no official release of CMSIS_6 yet. So you could either clone the repo(s) and add them locally via cpackget... or you can use the devdrop releases described at https://github.com/ARM-software/CMSIS_6/releases/tag/dev%2Fv6.0.0-dev53

tldr; Download the [packlist.txt](https://github.com/ARM-software/CMSIS_6/releases/download/dev%2Fv6.0.0-dev53/packlist.txt) and run `cpackget add -f packlist.txt`