ARM-software/CMSIS_5 #797: Can gated recurrent units be deployed on ARM cortex a53?

Pradeepcbk, 2020-01-14:
Hello All,

I have a task of performing regression learning on a target board (ARM Cortex A53 processor).

I have completed the tasks of data and feature extraction.

Now, i am analyzing the performances of different learning algorithms (GRUs and LSTM) in tensorflow.

However, my concern is that not all models has the software supports to be deployed on the target board.

Out of the software libraries privided by ARM for ML models, i found out that only ARM NN on top of ARM compute library is the way to go. However, in the description of the compute library - it specifies that it supports only SVM (Binary Classification) and CNN models.

I wanted to know, if there is any other way of accomplishing this task? 

Thanks in advance

felix-johnny, 2020-04-06:
@gmiodice Could you answer this ACL related question?

gmiodice, 2020-04-06:
Hi @Pradeepcbk ,

and thanks for using our SW libraries :) .

So in ArmNN/ACL we do support LSTM. If you are interested on running a network with the LSTM operator, you can find an example in ACL under the folder "examples" called [graph_deepspeech_v0_4_1.cpp](https://github.com/ARM-software/ComputeLibrary/blob/master/examples/graph_deepspeech_v0_4_1.cpp).

Let me know if you need any further information about that.
Thanks,
Gian Marco

closed 2020-06-10 by felix-johnny