ARM-software/CMSIS_5 #798: Code size of RTX too big more than 12KB

marduino, 2020-01-14:
I am working on a CM3 based project, my ported RTX with CMSIS-5 seems too big, over 12KB '.text'

it is basically working,  I did some tailor, now the code size is a big issue to me,  according with http://www.keil.com/support/man/docs/rlarm/rlarm_ar_technical_data.htm

It says the code size is <4KB for  cortex-M? Is there anything I do wrong? 

[image]

JonatanAntoni, 2020-01-14:
Hi @marduino,

this link belongs to an old version from RTX, not the current RTX5 one that is still under maintenance and shipped with MDK 5. Unfortunately we don't have specific memory requirements measures for RTX5 because it highly depends on the used Compiler and optimization level.

You could try to open one of our very basic Blinky examples and compare different settings. Does this help?

Cheers,
Jonatan

marduino, 2020-01-14:
@JonatanAntoni I can't use my tool to analysis in details, but I got even bigger code space, FYI.

```
    Total RO  Size (Code + RO Data)                11844 (  11.57kB)
    Total RW  Size (RW Data + ZI Data)              7400 (   7.23kB)
    Total ROM Size (Code + RO Data + RW Data)      11900 (  11.62kB)
``` 

I attach the Blinky map file for your reference
[Blinky.map.txt](https://github.com/ARM-software/CMSIS_5/files/4058562/Blinky.map.txt)

marduino, 2020-01-14:
FYI, this is compiled with highest optimization level 'O3'

JonatanAntoni, 2020-01-14:
Well, the total size is not only about RTX5 itself but it includes Event Recorder and other stuff.

marduino, 2020-01-15:
@JonatanAntoni you are right, sorry for my laziness, just adjust removed Event Recorder. The total size now is little better than mine, still not as good as I expected.

Now the summarize is:
```
    Total RO  Size (Code + RO Data)                 8464 (   8.27kB)
    Total RW  Size (RW Data + ZI Data)              6184 (   6.04kB)
    Total ROM Size (Code + RO Data + RW Data)       8520 (   8.32kB)

```
Here is detail:
```
       510        106          0         20         16      14044   blinky.o
       208         12          0          0          0        636   irq_cm4f.o
       118         20          0          1          0       2444   os_systick.o
         6          0          0          0          0       1768   retarget_io.o
         4          0          0          0          0       1009   rtx_config.o
        82          4          0          0          0      21769   rtx_delay.o
       608         50         11        164          0      22622   rtx_kernel.o
         2          0        260       4940          0       9478   rtx_lib.o
       210          0          0          0          0       3543   rtx_memory.o
       452          0          0          0          0       5445   rtx_mempool.o
      1614         38          0          0          0      22784   rtx_msgqueue.o
        70          0          0          0          0       1485   rtx_mutex.o
       624         30          0          0          0       8889   rtx_system.o
      2060        106          0          0          0      29471   rtx_thread.o
       212         24          0          0          0       2819   rtx_timer.o
        36          8        392          0       1024        864   startup_stm32l476xx.o
       236         28         64          4          0       2265   system_stm32l4xx.o

```

RTX kernel takes:
[image]

JonatanAntoni, 2020-03-26:
Hi @FlynnMa,

I can confirm that this is as expected. 

I just realized the http://www.keil.com/support/man/docs/rlarm/rlarm_ar_technical_data.htm points to values from RTX4, not RTX5.

Since RTX4 we added some features which need more code space.

Cheers,
Jonatan

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni