ARM-software/CMSIS_5 #802: SVDConv does not support more than 240 interrupts

simtind, 2020-01-20:
Cortex-M33 should support up to 480 interrupts, but when declaring an interrupt with interrupt number 240 or above, it triggers error M331, independent of the  value of `<deviceNumInterrupts>.`

*** ERROR M331: interrupts.svd (Line 35)
  Interrupt 'LAST' Number '240' greater 239.

*** INFO M211: interrupts.svd (Line 35)
  Ignoring Interrupt : 'LAST' (see previous message)

Attached is a minimal svd file that reproduces the issue.
[interrupts.txt](https://github.com/ARM-software/CMSIS_5/files/4085151/interrupts.txt)

JonatanAntoni, 2020-01-20:
Hi @simtind,

thanks for pointing this out. We'll have a look into that one.

Cheers,
Jonatan

simtind, 2020-09-11:
Hi, any movement on this issue?

simtind, 2021-08-13:
Hi @JonatanAntoni, 
Have you had time to look at this?

JonatanAntoni, 2021-09-06:
Hi @simtind,

sorry, this slipped through while we were working on getting the CMSIS-Pack code to open source.

Cheers,
Jonatan