ARM-software/CMSIS_5 #804: Enable integration with code quality tools

escherstair, 2020-01-21:
This request is partialli related to #536, which refers to a public CI infrastructure.
I understand that this is a huge topic and it will not be done (probably).
For this reason I suggest to invest a limited effort in integration of Code Quality tools.
This kind of tools can help a lot finding potential bugs and giving suggestions to the developers.
[SonarQube](https://www.sonarqube.org/), [Coverity](https://scan.coverity.com/), [Codacy](https://www.codacy.com/) and [LGTM](https://lgtm.com/) as an example are free of charge for open source project and are widely used on public github repositories.
As I wrote in #536 I've already enabled Codacy integration on my fork of CMSIS_5 and @JonatanAntoni wrote it looks promising.

I ask to some of CMSIS_5 mainteners to enable one (of many) of the above tools to add powerful tools to check the code quality of this important repository.

JonatanAntoni, 2020-01-21:
Can you give me some insights what you did to enable that? If this is not too much effort I should be able to enable the very same on this upstream repo.

escherstair, 2020-01-21:
You can follow [the steps](https://support.codacy.com/hc/en-us/articles/360010263540) on codacy website.
And then you can [add a badge](https://support.codacy.com/hc/en-us/articles/212799365-Badges) to CMSIS_5 repository.

JonatanAntoni, 2020-01-21:
Unfortunately, it is not that easy for a company hosted repository. I cannot get this approved without some efforts. I'll double check with the Codacy team but I doubt we won't be able to provide this.

JonatanAntoni, 2020-01-22:
Codacy requires a level of access on organization level which I cannot get approved, easily. We are running static code analysis with professional/commercial tools internally. Hence the additional benefits of integration yet another tool publicly is very limited. I cannot further support this request for the time being..

closed 2020-01-22 by JonatanAntoni