ARM-software/CMSIS_5 #807: RTX5 memory pool does not handle the double free properly

AoDotH, 2020-01-21:
The RTX5 (RTOS2) memory pool does not handle the double free properly. 
1. It does not return any error while double freeing the memory pool. This can lead to a hard fault later when an allocation of memory. 
2. Additionally, I have an example project where after double free, if we invoke the allocate, it kept on allocating the same buffer again and again.
Here is the project link:
https://github.com/AoDotH/CMSIS_RTX5_Issue/tree/master/MemPoolFreeIssue

JonatanAntoni, 2020-01-22:
Hi @AoDotH,

thanks for raising this. Unfortunately we don't specifically track if a block of memory is allocated or not. Hence we cannot detect a double free, easily.

We discussed possibilities to enhance that multiple time. So far we came to the conclusion that the performance impact would outweigh the benefits. Thus we don't have any plans to change this behavior.

The user needs to be careful not to cause double free's. A common approach and best practice is to set the pointer to `NULL` after having free'd its memory, e.g.
```
[..]
status = osMemoryPoolFree (memPoolId, tempPtr);
tempPtr = NULL;
[..]
```

Cheers,
Jonatan

AoDotH, 2020-01-25:
Thanks for the prompt reply @JonatanAntoni .
I will close the issue.

AoDotH, 2020-01-25:
The checks for double free of memory pool will be handled directly by the user and as from my experience it causes problem in subsequent allocation of memory, which is already documented in the osMemoryPoolFree API documentation.

closed 2020-01-25 by AoDotH