ARM-software/CMSIS_5 #814: version code in cmsis_version.h mismatch with release tag 5.6.0

FlynnMa, 2020-02-04:
I get code from https://github.com/ARM-software/CMSIS_5/releases/tag/5.6.0 , in which the source file cmsis_version.h includes bellow version code:

```
/*  CMSIS Version definitions */
#define __CM_CMSIS_VERSION_MAIN  ( 5U)                                      /*!< [31:16] CMSIS Core(M) main version */
#define __CM_CMSIS_VERSION_SUB   ( 3U)                                      /*!< [15:0]  CMSIS Core(M) sub version */
#define __CM_CMSIS_VERSION       ((__CM_CMSIS_VERSION_MAIN << 16U) | \
                                   __CM_CMSIS_VERSION_SUB           )       /*!< CMSIS Core(M) version number */
```

The file header also mismatch:
```
 * @file     cmsis_version.h
 * @brief    CMSIS Core(M) Version definitions
 * @version  V5.0.3
 * @date     24. June 2019
```

FlynnMa, 2020-02-04:
Personally I would suggest to create a tool to do that, so the only thing needs to do is to create a release tag, the cmsis_version.h will be generated automatically during compile time.

Not sure if I can create a pull request? If I can, I will be more than happy to make such auto generation script for a review. Thanks!

JonatanAntoni, 2020-02-04:
Hi @FlynnMa,

I appreciate that different versions can be misleading. We use independent versioning on multiple layers:
1. CMSIS Release Bundle, which is reflected by the tag
2. Software components in the bundle, e.g. this `cmsis_version.h` belongs to CMSIS-Core(M)
3. Versions and change dates of individual files, e.g. the specific `cmsis_version.h`

In the concrete case, CMSIS Release Bundle 5.6.0 contains CMSIS-Core(M) in version 5.3.0, and the `cmsis_version.h` file version is `5.0.3`. Three different versions.

We played through a lot of variants on how to cope with versions on different layers. The solution we use since a while seems to be the only practical one.

Cheers,
Jonatan

closed 2020-03-26 by JonatanAntoni

thebruce87m, 2020-03-31:
Tagging on to this since it is relevant:

I have a project with CMSIS already built in - how to I determine which version (Bundle?) I have from the source code?

JonatanAntoni, 2020-04-01:
Please find my answer to #878.