ARM-software/CMSIS_5 #818: SVDConv - ARM Compiler 6 C++ support

jritchey-renesas, 2020-02-05:
The structs generated by `SVDConv --fields=struct ...` define anonymous bit fields like the following:
 ~~~~
struct {
...
    __IO uint32_t BIT_FIELD : 6; /* BIT_FIELD */
    __I  uint32_t           : 2; /* Reserved */
...
} REG1;
~~~~

This generates the following error in C++ when compiled with ARM Compiler 6.
`error: anonymous bit-field cannot have qualifiers`

I didn't find any options for ARM Compiler 6 to 'mask' this error.

Would it be possible to add support to SVDConv to fill in a name for each field (Eg. 'Reserved1, Reserved2...), or remove the qualifiers so that this will compile?

JonatanAntoni, 2020-03-02:
Should be fixed in SVDConv 3.3.35, see 0b63cbb.
Grab Windows version from https://github.com/ARM-software/CMSIS_5/raw/develop/CMSIS/Utilities/Win32/SVDConv.exe

jritchey-renesas, 2020-03-03:
Yes that solves the issue.

Thanks!
-Jackson

closed 2020-03-03 by jritchey-renesas