ARM-software/CMSIS_5 #823: Question regarding SysTick_Handler implementation.

vijaykumarprathipati, 2020-02-10:
Hi,

In SysTick_Handler implementation for Arm Cortex M0 - R0 and LR is pushed to stack (https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/ARM/irq_cm0.s#L164). I understand that LR needs to be preserved as it holds EXC_RETURN but Is there any reason that R0 needs to be preserved as well ? Looks like the PendSV_Handler also doing the same.

Regards,
Vijay.

JonatanAntoni, 2020-02-10:
HI @vijaykumarprathipati,

this additional "dummy" register push/pop is made to keep the stack pointer 8-byte aligned according to the [Arm Architecture Procedure Call Standard (AAPCS)](https://developer.arm.com/docs/ihi0042/h).

Further readings:
https://community.arm.com/developer/ip-products/processors/f/cortex-m-forum/6344/what-is-the-meaning-of-a-64-bit-aligned-stack-pointer-address

Cheers,
Jonatan

vijaykumarprathipati, 2020-02-10:
Thanks @JonatanAntoni for Great Help.

closed 2020-02-10 by vijaykumarprathipati