ARM-software/CMSIS_5 #824: Use of export "C"

CurtisAMU, 2020-02-10:
While cmsis_os2.h and some RTX headers (e.g. rtx_os.h) contain the
#ifdef  __cplusplus
extern "C"
{
#endif
idiom, others do not. We're using CMSIS-RTOS2 in C++ projects where _export "C"_ in os_tick.h is required.
I think, it would be consistent to include export "C" in all header files that may be included by CMSIS users.

JonatanAntoni, 2020-02-10:
Hi @CurtisAMU,

thanks for pointing this out. You're right and we are going to add the `extern "C"` declaration to `os_tick.h`.
Please take care to compile the RTX source code itself in C mode (not C++).

Cheers,
Jonatan

RobertRostohar, 2021-05-17:
extern "C" has been added as requested.

JonatanAntoni, 2021-06-08:
@CurtisAMU, may I ask you to check the fix and close the issue, please?

CurtisAMU, 2021-06-10:
Thank you. I checked the fix and it works.

closed 2021-06-10 by CurtisAMU