ARM-software/CMSIS_5 #828: CMSIS DSP Bit reverse Index table generation

CairnGlen, 2020-02-14:
Hi
has anyone figured out the generation algorithm for  the following tables

const uint16_t armBitRevIndexTable_fixed_4096[ARMBITREVINDEXTABLE_FIXED_4096_TABLE_LENGTH] =
/* radix 4, size 4032 */
    8,16384, 16,8192, 24,24576, 32,4096, 40,20480, 48,12288, 56,28672, 64,2048,
  .......

const uint16_t armBitRevIndexTable4096[ARMBITREVINDEXTABLE_4096_TABLE_LENGTH] =
/* radix 8, size 4032 */
   8,4096, 16,8192, 24,12288, 32,16384, 40,20480, 48,24576, 56,28672, 64,512,

I have been able to figure out a pattern for the first 64 elements but it fails after that

Thank you
CG

christophe0606, 2020-03-11:
I have created this new issue to explain how to do it. I hope it answers your question:
https://github.com/ARM-software/CMSIS_5/issues/858

CairnGlen, 2020-03-11:
Perfect thank you Christophe

closed 2020-03-11 by CairnGlen