ARM-software/CMSIS_5 #840: A lot of conversion and sign-conversion warnings

jankap, 2020-02-25:
There are a lot of warnings, especially in the DSP when compiled with `armclang` version 6.10.1 with these options `-g3 -Wall  -Wextra  -Wshadow -Wdouble-promotion -Wformat=2  -Wundef -fno-common  -Wconversion -ffunction-sections -fdata-sections` 

Most of them are implicit conversion warnings like
```
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ControllerFunctions/ControllerFunctions.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ControllerFunctions/arm_pid_init_f32.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(652): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t) (__CLZ( in) - 1));
                              ~~~~~  ^~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(656): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t) (__CLZ(-in) - 1));
                              ~~~~~ ^~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(703): warning: implicit conversion changes signedness: 'q15_t' (aka 'short') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t)(__CLZ( in) - 17));
                             ~~~~~  ^~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(707): warning: implicit conversion changes signedness: 'int' to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t)(__CLZ(-in) - 17));
                             ~~~~~ ^~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(711): warning: implicit conversion loses integer precision: 'int' to 'q15_t' (aka 'short') [-Wconversion]
    in = (in << signBits);
       ~  ~~~^~~~~~~~~~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(727): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'unsigned int' [-Wsign-conversion]
      out = (q15_t) (((q31_t) out * tempVal) >> 14);
                      ^~~~~~~~~~~ ~
6 warnings generated.
compiling ControllerFunctions.c...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ComplexMathFunctions/ComplexMathFunctions.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_conj_f32.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(652): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t) (__CLZ( in) - 1));
                              ~~~~~  ^~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(656): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t) (__CLZ(-in) - 1));
                              ~~~~~ ^~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(703): warning: implicit conversion changes signedness: 'q15_t' (aka 'short') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t)(__CLZ( in) - 17));
                             ~~~~~  ^~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(707): warning: implicit conversion changes signedness: 'int' to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t)(__CLZ(-in) - 17));
                             ~~~~~ ^~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(711): warning: implicit conversion loses integer precision: 'int' to 'q15_t' (aka 'short') [-Wconversion]
    in = (in << signBits);
       ~  ~~~^~~~~~~~~~~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(727): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'unsigned int' [-Wsign-conversion]
      out = (q15_t) (((q31_t) out * tempVal) >> 14);
                      ^~~~~~~~~~~ ~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ComplexMathFunctions/ComplexMathFunctions.c(30): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_conj_q15.c(146): warning: implicit conversion loses integer precision: 'q31_t' (aka 'int') to 'q15_t' (aka 'short') [-Wconversion]
    *pDst++ = (in1 == (q15_t) 0x8000) ? (q15_t) 0x7fff : -in1;
            ~                                            ^~~~
7 warnings generated.
compiling ComplexMathFunctions.c...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/BasicMathFunctions/BasicMathFunctions.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f32.c(29): warning: In file included from...
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(652): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
      signBits = ((uint32_t) (__CLZ( in) - 1));
                              ~~~~~  ^~
C:/Keil_v5/ARM/PACK/ARM/CMSIS/5.5.1/CMSIS/DSP/Include\arm_math.h(656): warning: implicit conversion changes signedness: 'q31_t' (aka 'int') to 'uint32_t' (aka 'unsigned int') [-Wsign-conversion]
```

Are there any plans to fix that or are the compiler settings too aggressive?

Thanks,
Jan

christophe0606, 2020-02-27:
I generally only use -Wall -Wextra but indeed -Wconversion is showing a lot more warnings.

And even if it is making the code more verbose, I think those warnings should be removed.
It is better to have the conversion explicit in the code.

So, I'll do the fix but since it is impacting lot of files and everything will have to be validated again, it will take some time.

christophe0606, 2020-04-14:
I have made the exercise to remove all of those warnings.
It is requiring many many changes and is making the code less readable with lot of typecasts.

I am a bit reluctant at doing so many changes on the code because it will require a big re-validation effort.

So, it is a change we are going to do progressively release after release to avoid taking any risk.

In all the warnings, I have seen no serious one. Most are coming from the use of intrinsics.

There are a few cases where it would indeed be better to make the conversion or sign-conversion explicit.

As consequence, I am switching the status to "Future" instead of "Enhancement".

escherstair, 2022-05-16:
Hello @christophe0606 
is there any update on this task?
All of these warnings should be handled during a certfication of the source code.

christophe0606, 2022-05-16:
Hello @escherstair , unfortunately no update on this task.

escherstair, 2022-05-16:
Thanks @christophe0606 
I understand that fixing all of them is a huge job, but I think at least the more "suspicious" ones should be handled.
Do you agree if I link here these ones?

christophe0606, 2022-05-16:
@escherstair Yes please. List the most suspicious ones. It will be useful when I can find some bandwidth to work on it.

escherstair, 2022-05-16:
This one:
`CMSIS/DSP/Include/dsp/utils.h(219): warning: operand of ? changes signedness: 'q63_t' (aka 'long long') to 'uint64_t' (aka 'unsigned long long') [-Wsign-conversion]
Because here
`https://github.com/ARM-software/CMSIS_5/blob/d252f7a4061155abaa1c9817069eed8888c79543/CMSIS/DSP/Include/dsp/utils.h#L219
this kind of implicit cast can be dangerous in the "corner cases" (`num = 0x8000000` as a `q31_t`)

This one:
`CMSIS/DSP/Include/dsp/utils.h(204): warning: implicit conversion changes signedness: 'unsigned int' to 'int32_t' (aka 'int') [-Wsign-conversion]`
Because here
https://github.com/ARM-software/CMSIS_5/blob/d252f7a4061155abaa1c9817069eed8888c79543/CMSIS/DSP/Include/dsp/utils.h#L204
I can see a mess with the sign of the operands.

Basically, all of the warning are inside [utils.h](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/dsp/utils.h)

escherstair, 2022-05-16:
As far as I understand, many of the warnings come from casting a `signed` value to an `unsigned` value of the same zise (i.e. `q31_t` to `uint32_t`, and so on).
This is "safe" as @christophe0606 wrote in some comment above only in two’s-complement C implementations, which are the most common.
In this cases, this kind of cast doesn't alter the bit representation of the value.
But this is not guaranteed by the C standard:
the cast `(unsigned int)` a is not defined to retain the bits, but (C11 6.3.1.3p2) says

> the value is converted by repeatedly adding or subtracting one more than the maximum value that can be represented in the new type until the value is in the range of the new type.

This is one of the reasons towards the handling of the warnings.

christophe0606, 2022-05-16:
@escherstair I agree those warnings must be corrected at some point. There is lot of legacy around it so it is impacting lot of files. I'll start with the "new" functions you have highlighted

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606