ARM-software/CMSIS_5 #860: Question: RTX Object Identifiers.

vijaykumarprathipati, 2020-03-12:
Hello Everyone,

I have been going through CMSIS RTX code and came across the following macro definitions.

/// Object Identifier definitions
#define osRtxIdInvalid                0x00U
#define osRtxIdThread                0xF1U
#define osRtxIdTimer                  0xF2U
#define osRtxIdEventFlags          0xF3U
#define osRtxIdMutex                 0xF5U
#define osRtxIdSemaphore         0xF6U
#define osRtxIdMemoryPool       0xF7U
#define osRtxIdMessage              0xF9U
#define osRtxIdMessageQueue   0xFAU

I understand that this identifiers protects appropriate data structures by allowing us to check whether a structure has been clobbered/ a wrong structure is passed to the routine.  My question is that "what pattern is followed while giving values for these identifiers ?" In the above code 0xF4 & 0xF8 are omitted. So, If someone wants to add a new object(I don't see a reason to add though), what value should be used ?

Regards,
Vijay.

RobertRostohar, 2020-03-12:
Those identifiers are RTX internals and there should be no reason for the user to change them or add new ones.

Regardless of that, the values 0xF4 and 0xF8 have been omitted to avoid memory pool free side effect.

Memory pools use the first 32-bits as the address to the next free block when a block is released. 

When object memory is allocated from dynamic memory pools and a block is released then the object id overlaps with the 32-bit address (also aligned to 32-bit). In order to avoid a valid object id derived from a 32-bit address the identifiers are chosen to have the 2 LSB cleared to 0.

vijaykumarprathipati, 2020-03-13:
> Those identifiers are RTX internals and there should be no reason for the user to change them or add new ones.
I totally agree with that. I am just trying to understand RTX internals as a case study.
Anyways, Thanks for the valuable information. IMHO, these information would be more helpful if the same is presented in comments at the corresponding code. I spent nearly 2 hours trying to understand the pattern for these values but couldn't get any, So I reached to you. It would be a timesaver for those who love CMSIS RTX and trying to explore it.

vijaykumarprathipati, 2020-03-13:
Also, Is there any reason why identifiers MSB are all chosen to 1 ?

RobertRostohar, 2020-03-16:
Upper nibble id identifiers are all 1s for the same reason however in case for big-endian devices. Normal R/W memory would never be at address starting with 0xF…

vijaykumarprathipati, 2020-03-19:
@RobertRostohar - Thanks again for the Information.

closed 2020-03-19 by vijaykumarprathipati