ARM-software/CMSIS_5 #861: unaligned 4 bytes access is replaced by memcpy, adds too much overhead

RockySong, 2020-03-16:
"__SIMD32" was replaced by "arm_nn_read_q15x2_ia" in order to read 4 bytes from unaligned address. Though this guarantees no unaligned fault due to generated LDR/LDM instructions in ac6 (ac5 does not), its implementation is memcpy() which adds too much overhead.
In fact most use cases can accept unaligned access, suggest to create a new configuration that determines whether to allow unaligned access and reminds user to enable it if they want better performance.

felix-johnny, 2020-04-06:
@RockySong  Sorry for the delay. Was it with GCC or ARMC6 that you observe the overhead? We did have some reports of overhead using the memcpy(https://github.com/tensorflow/tensorflow/issues/37361) ..Do you see it without -fno-builtin ?

felix-johnny, 2020-04-06:
And this is the background to the issue https://github.com/ARM-software/CMSIS_5/issues/546

RockySong, 2020-04-09:
I see a call to aeabi memcpy

felix-johnny, 2020-04-14:
@RockySong Not sure that clarifies the question. Have you tied without the  -fno-builtin option? I believe that should resolve the issue that you see.

felix-johnny, 2020-06-10:
I am closing this issue and if your issue is not resolved after using -fno-builtin , please re open it.

closed 2020-06-10 by felix-johnny