ARM-software/CMSIS_5 #866: Does CMSIS-DAP support the float64 matrix operations?

WangShanpeng, 2020-03-19:
https://github.com/ARM-software/CMSIS_5/tree/master/CMSIS/DSP/Source/MatrixFunctions
I see that there are only `arm_mat_mult_f32`, `arm_mat_add_f32` functions in the source code. 
Does CMSIS-DAP support the float64 matrix operations?

christophe0606, 2020-03-19:
We have started to add float64 versions of the functions but right now for matrix there is only the inverse.

WangShanpeng, 2020-03-19:
> We have started to add float64 versions of the functions but right now for matrix there is only the inverse.

Thanks very much for your reply. I would like to know how soon the float64 version will be released?

christophe0606, 2020-03-19:
I unfortunately don't know.

WangShanpeng, 2020-03-19:
@christophe0606 OK, thank you!

closed 2020-03-19 by WangShanpeng