ARM-software/CMSIS_5 #870: Tables for arm_rfft_fast_sR_f32_len32 structure are incorrect.

caseymillsdavis, 2020-03-20:
I made this patch locally some weeks ago and I believe it fixed incorrect results I was seeing with length 32 real ffts. Certainly what is there looks incorrect in that the sizes for the twiddle coefficients and reverse index table are not analogous for the other definitions that follow.

```
diff --git a/CMSIS/DSP/Source/CommonTables/arm_const_structs.c b/CMSIS/DSP/Source/CommonTables/arm_const_structs.c
index 166e8cef..6b30c743 100644
--- a/CMSIS/DSP/Source/CommonTables/arm_const_structs.c
+++ b/CMSIS/DSP/Source/CommonTables/arm_const_structs.c
@@ -359,7 +359,7 @@ const arm_rfft_fast_instance_f64 arm_rfft_fast_sR_f64_len4096 = {
 
 #if !defined(ARM_DSP_CONFIG_TABLES) || defined(ARM_ALL_FFT_TABLES) || (defined(ARM_TABLE_TWIDDLECOEF_F32_32) && defined(ARM_TABLE_BITREVIDX_FLT_32) && defined(ARM_TABLE_TWIDDLECOEF_RFFT_F32_32))
 const arm_rfft_fast_instance_f32 arm_rfft_fast_sR_f32_len32 = {
-  { 16, twiddleCoef_32, armBitRevIndexTable32, ARMBITREVINDEXTABLE_16_TABLE_LENGTH },
+  { 16, twiddleCoef_16, armBitRevIndexTable16, ARMBITREVINDEXTABLE_16_TABLE_LENGTH },
   32U,
   (float32_t *)twiddleCoef_rfft_32
 };
```

christophe0606, 2020-03-23:
Thanks for reporting this.
I have just corrected it in commit https://github.com/ARM-software/CMSIS_5/commit/1a98b761ba7e07e2491f49c094d3cd85bbf23521

Wastus, 2020-05-18:
Shouldn't the #ifdef be changed accordingly?

`armBitRevIndexTable16` is only there if `ARM_TABLE_BITREVIDX_FLT_16` is defined, but not when `ARM_TABLE_BITREVIDX_FLT_32` is defined.

(same goes for all the other rfft structs.

Wastus, 2020-05-18:
If my understanding is correct the `arm_const_structs.c` should give us constant structures so we don't need to call the init function from `arm_rfft_fast_init_f32.c`.
So the requirement to have the constant structure should be the same for calling the init function.

from `arm_rfft_fast_init_f32.c`:
```
#if !defined(ARM_DSP_CONFIG_TABLES) || defined(ARM_ALL_FFT_TABLES) || (defined(ARM_TABLE_TWIDDLECOEF_F32_16) && defined(ARM_TABLE_BITREVIDX_FLT_16) && defined(ARM_TABLE_TWIDDLECOEF_F32_16) && defined(ARM_TABLE_TWIDDLECOEF_RFFT_F32_32))
/**
...
 */
arm_status arm_rfft_32_fast_init_f32( arm_rfft_fast_instance_f32 * S ) {

  arm_cfft_instance_f32 * Sint;

  if( !S ) return ARM_MATH_ARGUMENT_ERROR;

  Sint = &(S->Sint);
  Sint->fftLen = 16U;
  S->fftLenRFFT = 32U;

  Sint->bitRevLength = ARMBITREVINDEXTABLE_16_TABLE_LENGTH;
  Sint->pBitRevTable = (uint16_t *)armBitRevIndexTable16;
  Sint->pTwiddle     = (float32_t *) twiddleCoef_16;
  S->pTwiddleRFFT    = (float32_t *) twiddleCoef_rfft_32;

  return ARM_MATH_SUCCESS;
}
}
#endif
```

compared to `arm_const_structs.c`:
```
#if !defined(ARM_DSP_CONFIG_TABLES) || defined(ARM_ALL_FFT_TABLES) || (defined(ARM_TABLE_TWIDDLECOEF_F32_32) && defined(ARM_TABLE_BITREVIDX_FLT_32) && defined(ARM_TABLE_TWIDDLECOEF_RFFT_F32_32))
const arm_rfft_fast_instance_f32 arm_rfft_fast_sR_f32_len32 = {
  { 16, twiddleCoef_16, armBitRevIndexTable16, ARMBITREVINDEXTABLE_16_TABLE_LENGTH },
  32U,
  (float32_t *)twiddleCoef_rfft_32
};
#endif 
```
The #if defined should probably be the same.
In `arm_rfft_fast_init_f32.c` the second `ARM_TABLE_TWIDDLECOEF_F32_16` could be removed for clarity. But in `arm_const_structs.c` it should be `ARM_TABLE_TWIDDLECOEF_F32_16` instead of `ARM_TABLE_TWIDDLECOEF_F32_32`.

christophe0606, 2020-05-29:
It should be corrected in commit https://github.com/ARM-software/CMSIS_5/commit/512aeff7bf8aebf026da3836d9830bba87822b8e

Thank for your report.

closed 2021-06-14 by christophe0606