ARM-software/CMSIS_5 #875: CMSIS-Build

jankap, 2020-03-27:
Dear all,

is the development of CMSIS-Build open source or will it be open sourced? We're extremely much looking forward to have a way to trigger builds, based on CMSIS-Pack and custom code by IDE (uVision or Eclipse), as well as by (c)make. Did I understand that right that this is a possible solution?

Thank you very much

jkrech, 2020-03-30:
Hello,
at the heart of CMSIS-Build is the command line tool 'cmsisbuildgen' which is currently still under development. Similar to the tools like SVDConv and PackChk we currently plan to deliver this tool as binary for Windows and Linux under the Apache 2.0 license.

Best regards,

Joachim

jankap, 2020-03-30:
Thank you. That means, no sources will be available? To be hones, I was hoping to get a better understanding of all the different file formats.

Will there be a conversion tool for Keil/ARM .uvprojx files? In the pdf announcing CMSIS-Build, (c)make is mentioned. Right now it's a huge pain to compile Keil projects with make. We need both.

Thanks.

jkrech, 2020-03-30:
Hello,
MDK 5.30 targeted for e/o April 2020 does include an export and import feature for the new CPRJ CMSIS-Build project format.
The latest format specification can be found here: 
https://arm-software.github.io/CMSIS_5/Build/html/element_cprj.html

cmsisbuildgen at this point only supports 'make' not cmake.

~ Joachim

loverdeg, 2020-08-18:
Open up development

brondani, 2021-09-23:
Hi @jankap,
Now CMSIS-Build is open source and it generates `CMakeLists`, this is the repository:
https://github.com/Open-CMSIS-Pack/devtools
Here you can find the latest release:
https://github.com/Open-CMSIS-Pack/devtools/releases/tag/tools%2Fbuildmgr%2F0.10.3
Can you please check whether it meets your expectations?
Thanks
Daniel

jankap, 2021-10-14:
Yes, great news. Thank you very much!