ARM-software/CMSIS_5 #876: FFT slower with dsp instructions (gcc)

rosek86, 2020-03-28:
Hi,

I was facing a problem in FFT calculations time (Q15), the version with ```-U__ARM_FEATURE_DSP``` was executing much faster than the version with DSP instructions enabled. For 16k samples, I got ~40 ms and ~200 ms execution time respectively (O3).

It turned out that the problem was related to ```read_q15x2```-like inline functions, they use ```memcpy``` so the compiler generates a lot of ```bl``` calls.

I rewrote these functions to use pointers instead of ```memcpy```, and the execution time dropped to about 20 ms.

```C
__STATIC_FORCEINLINE q31_t read_q15x2 (q15_t * pQ15)
{
 return *(uint32_t *)pQ15;
}
```

Is this expected behaviour, is it possible to fix this issue without modifying CMSIS library, please?

gcc: 7.3.1, 9.2.1
mcu: Cortex-M4 (nRF52840)

christophe0606, 2020-03-30:
@rosek86 It is a problem coming back quite often.
We did the change because the version with pointers (and it was written like that in previous version) is not correct C (according to our C compiler developers) and is indeed giving wrong code with some compilers.

Are you using -fno-builtin when building with gcc ?

GuentherMartin, 2020-03-30:
The performance depends on how the used compiler optimizes the code.
e.g.: the ARM compiler lowers the memcpy to just load and store instructions because  only small amounts of data are memcpy'ed. With a higher optimization level there are no function calls to memcpy any longer.
I can not tell how GCC optimizes this kind of function calls.
If you access the data via casted pointers you mast take care that the data is always aligned otherwise you can get an "unaligned access" exception.

rosek86, 2020-03-30:
Hello all,

Thank you very much @christophe0606. You are right that I've been using -fno-builtin, which was a copy from Nordic Semiconductor's SDK. I get expected 20 ms once I remove this flag, this is the same as for the pointers version.

closed 2020-03-30 by rosek86

rosek86, 2020-03-31:
@GuentherMartin out of curiosity does the memcpy protect from the unalignment pointers if compiler optimizes out the call, please?

Ref. http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka3934.html

"Due to these optimizations, you must take care when copying data using unaligned pointers. The ARM compiler assumes that all pointers are naturally-aligned"

GuentherMartin, 2020-03-31:
Hello @rosek86, thank you for pointing out.
We had a really long discussion about the "pointer access" with the compiler team and the outcome was to use "memcpy". I did not check every situation with ARM and GCC compiler but I trust the compiler team.

rosek86, 2020-03-31:
Understood, thanks for the response.