ARM-software/CMSIS_5 #878: How to determine CMSIS version from source code?

thebruce87m, 2020-04-01:
I had originally commented on #814 but I think this is really a separate issue.

As per #814 I am a little confused as to how to find out the version of CMSIS.

As I understand it you call a whole release a "Release Bundle" and give it a version number, e.g. 5.6.0. This number relates directly to the tagged versions on github.

If I have an existing project, for example a project generated from a tool like STM32CubeMX, how do you determine which version (Release Bundle?) of CMSIS is included in the project purely from the source code of CMSIS in the project?

I was hoping this would be in a header file such as **cmsis_version.h**, however **cmsis_version.h** as stated in #814 seems to relate to something else.

JonatanAntoni, 2020-04-01:
Hi @thebruce87m,

It depends on what you are looking for. The CMSIS release bundle (aka the pack) is just a collection of loosely coupled software components (and documentation, examples, aso.).

We don't have provisions to check the pack version from inside your application software. This might be misleading because basically you could combine different components from different pack versions.

Each software component has a specific version as well, such as in CMSIS 5.7.0 there are CMSIS-Core(M) Version 5.3.0 and RTX5 Version 5.5.1. I assume you are actually looking for these versions. For CMSIS-Core(M) there are some version defines you can use in `cmsis_version.h`. The version of RTX5 can be queried using the function `osKernelGetInfo`.

If you want to now which pack version these specific components have been shipped with, you'd need to refer to the release notes of the pack. The [revision history of CMSIS](https://arm-software.github.io/CMSIS_5/General/html/cm_revisionHistory.html) gives you a full list of pack versions and contained component versions. Please note that the very same component version might be part of different pack versions, e.g. if a component was not changed between to pack releases.

Does this answer your question?

Cheers,
Jonatan

thebruce87m, 2020-04-01:
Hi @JonatanAntoni 

Thanks for your answer - it helps greatly. I do have an additional question based on what I am seeing. I know this is kind of related to how STM32CubeMX is doing things, and therefore not really your concern but I would appreciate your insight and I think a lot of people will be in the same boat.

So, my project as generated from STM32CubeMX has a CMSIS folder that looks as follows:

> .
> ├── Device
> │   └── ST
> │       └── STM32L4xx
> │           └── Include
> │               ├── stm32l4s7xx.h
> │               ├── stm32l4xx.h
> │               └── system_stm32l4xx.h
> └── Include
>     ├── cmsis_armcc.h
>     ├── cmsis_armclang.h
>     ├── cmsis_compiler.h
>     ├── cmsis_gcc.h
>     ├── cmsis_iccarm.h
>     ├── cmsis_version.h
>     ├── core_armv8mbl.h
>     ├── core_armv8mml.h
>     ├── core_cm0.h
>     ├── core_cm0plus.h
>     ├── core_cm1.h
>     ├── core_cm23.h
>     ├── core_cm33.h
>     ├── core_cm3.h
>     ├── core_cm4.h
>     ├── core_cm7.h
>     ├── core_sc000.h
>     ├── core_sc300.h
>     ├── mpu_armv7.h
>     ├── mpu_armv8.h
>     └── tz_context.h

The folders structure **CMSIS/Device/ST/STM32L4xx/Include** seems to be vendor specific and not part of the CMSIS project,.

The folder structure **CMSIS/Include** seems to match the folder **/CMSIS_5/CMSIS/Core/Include** from the CMSIS github repo.

The table [here](https://arm-software.github.io/CMSIS_5/Core/html/index.html) says the following files are relevant for CMSIS-Core (Cortex-M):

File/Folder | Content
-- | --
CMSIS\Documentation\Core | This documentation
CMSIS\Core\Include | CMSIS-Core (Cortex-M) header files (for example core_cm3.h, core_cmInstr.h, etc.)
Device | Arm reference implementations of Cortex-M devices
Device\_Template_Vendor | CMSIS-Core Device Templates for extension by silicon vendors

and my cmsis_version.h is as follows:

```
/**************************************************************************//**
 * @file     cmsis_version.h
 * @brief    CMSIS Core(M) Version definitions
 * @version  V5.0.2
 * @date     19. April 2017
 ******************************************************************************/
/*
 * Copyright (c) 2009-2017 ARM Limited. All rights reserved.
 *
 * SPDX-License-Identifier: Apache-2.0
 *
 * Licensed under the Apache License, Version 2.0 (the License); you may
 * not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an AS IS BASIS, WITHOUT
 * WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#if   defined ( __ICCARM__ )
  #pragma system_include         /* treat file as system include file for MISRA check */
#elif defined (__clang__)
  #pragma clang system_header   /* treat file as system include file */
#endif

#ifndef __CMSIS_VERSION_H
#define __CMSIS_VERSION_H

/*  CMSIS Version definitions */
#define __CM_CMSIS_VERSION_MAIN  ( 5U)                                      /*!< [31:16] CMSIS Core(M) main version */
#define __CM_CMSIS_VERSION_SUB   ( 1U)                                      /*!< [15:0]  CMSIS Core(M) sub version */
#define __CM_CMSIS_VERSION       ((__CM_CMSIS_VERSION_MAIN << 16U) | \
                                   __CM_CMSIS_VERSION_SUB           )       /*!< CMSIS Core(M) version number */
#endif 
```

So my questions are:

* I think my project only uses the CMSIS-Core (Cortex-M) component, correct?
* The CMSIS-Core (Cortex-M) page [here](https://arm-software.github.io/CMSIS_5/Core/html/index.html) uses semver style MAJOR.MINOR.PATCH numbering, however the **cmsis_version.h** file only seems to have __CM_CMSIS_VERSION_MAIN and __CM_CMSIS_VERSION_SUB - does this equate to MAIN.SUB.0?
* I think my project therefore uses version 5.1.0 of CMSIS-Core (Cortex-M), correct?

Sorry it's a bit long winded, I want to be sure I understand it correctly.

Thanks again!

JonatanAntoni, 2020-04-01:
Hi @thebruce87m,

The bad news first: You got a copy of CMSIS instead of properly using CMSIS-Core(M) as an external dependency. So I can not even tell you from which bundle release this version has been taken.

The good thing is, you at least know that you have something around CMSIS-Core(M) 5.1.x which is a couple of years old. Unfortunately we don't track the patch revision at that point. So x might be 0, 1, or 2; which would be equivalent to CMSIS bundles 5.2, 5.3 or 5.4.

We'd need to take a much closer look to individual files to figure out the exact patch version. The version defines inside the code were not really intended to exactly identify the component version. The were rather intended to check for certain component features. And as we agree on semver adding features would result in an increased sub version. Patch versions should be feature equivalent. Hence no need to know the patch version.

Basically, I strongly recommend not to include dependencies this way, i.e. by copying the files into the project. The CMSIS-Pack concept has been designed to add use dependencies by reference. This makes version tracking and updating software components much easier because it can be automatically sorted by IDE or build tools.

Cheers,
Jonatan

thebruce87m, 2020-04-02:
Great, thanks!

closed 2020-04-02 by thebruce87m