ARM-software/CMSIS_5 #879: Unable to built Cmsis 5.7

sindhujaraga, 2020-04-01:
I have tried using CMSIS 5.7 package, in Atollic true studio but when I tried building it I have got an error  in  cmsis_gcc.h saying asm operand has impossible constraints error  and the error was at the line
 #define __ASM          __asm

GuentherMartin, 2020-04-01:
@sindhujaraga , could you please post the complete error message.
The define does surely not generate an error but maybe the embedded assembler where __ASM is used.

sindhujaraga, 2020-04-01:
**@GuentherMartin**  
__ASM ("sxtb16 %0, %1, ROR %2" : "=r" (result) : "r" (op1), "i" (rotate) )

CMSIS_5-develop/CMSIS/Core/Include/cmsis_gcc.h:41:50: error: impossible constraint in 'asm'
I am using stm32f746Ng mcu
I have used  CMSIS 5.6 didn't encounter this problem

thanks for the reply.

GuentherMartin, 2020-04-01:
This is curious because the embedded assembler part belongs to the compiler specific intrinsic  function __SXTB16_RORn. This function did not exist in CMSIS V5.6.0.
Do you use CMSIS components like DSP or NN?

sindhujaraga, 2020-04-01:
@GuentherMartin 
I am trying to use CMSIS DSP,NN

gmiodice, 2020-04-01:
Hi @sindhujaraga, 
could you please report the compiler version (GCC) used to compile CMSIS DSP,NN?
Many thanks in advance.

sindhujaraga, 2020-04-01:
gnu11(C11+gnu extensions)

JonatanAntoni, 2020-04-01:
@sindhujaraga,
Run `gcc --version` and report the full output, please.

valer23, 2020-06-19:
Hello there! How this error has been fixed? I encountered the same..
Here is my output of `gcc --version`:
```
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Thank you!

JonatanAntoni, 2020-06-19:
Hi @valer23,

no, as you can see this has silently be closed by the original requester.
Can you please share more details? What are you trying to build? Which error do you get?

Cheers,
Jonatan

JonatanAntoni, 2020-06-22:
Hello @valer23,

Looking closer to the GCC version you reported reveals you are using a host compiler not a target one, right? The inline assembly causing the error requires Arm target and does not work for x86.

If you are trying to compile CMSIS NN/DSP for your host machine (x86) you'd need to supply a different implementation for those defines.

cc: @christophe0606 can you comment on this?

Cheers,
Jonatan

GuentherMartin, 2020-06-22:
Hello @valer23,
when calling __SXTB16_RORn(uint32_t op1, uint32_t rotate) is the 2nd parameter a constant with value of 0, 8, 16 or 24?

christophe0606, 2020-06-22:
@JonatanAntoni  __SXTB16_RORn was introduced by NN team.
Looking at it quickly, it indeeds looks like the 2nd argument should be a compile time constant.

@gmiodice may help.

I have not seen any C fallback of this one for compiling for host.

valer23, 2020-06-22:
Hello @GuentherMartin, I don't directly call this macro. It is part of `arm_nn_mat_mult_nt_t_s8.c` file and it seems to call the macro with only the value 8.

@JonatanAntoni ok, so I should set my compiler according to my target and this will solve my problem? Which version of GCC should be used for this?

Thank you all.

GuentherMartin, 2020-06-22:
Hello @valer23,
thank you for your answer, I just want to know if the second parameter of the function call is a constant.

JonatanAntoni, 2020-06-22:
Hi @valer23,

the toolchains used for validating CMSIS-Core(M) are listed at https://arm-software.github.io/CMSIS_5/Core/html/index.html#tested_tools_sec

Basically all GCC versions with proper C99 support for Arm 32-bit Architectures should work (i.e. `--target=arm-arm-none-eabi`).

Cheers,
Jonatan

valer23, 2020-06-22:
@GuentherMartin yes the second parameter is a constant and is 8 at each call.
@JonatanAntoni the toolchains I use is already `arm-none-eabi-gcc` with architecture `ARM AArch32` ARM family is `cortex-m4`.
The toolchains is the one from `GNU MCU Eclipse ARM Embedded GCC (arm-none-eabi-gcc)`

JonatanAntoni, 2020-06-22:
Hi @valer23,

this should be fine. That's just not what you stated above.
Using this toolchain CMSIS-NN should compile without errors.

Cheers,
Jonatan

valer23, 2020-06-22:
Yes sorry the `gcc --version` I mentioned above was from terminal out of Eclipse environment. But my toolchains is `GNU MCU Eclipse ARM Embedded GCC (arm-none-eabi-gcc)` using gcc 9.2.1-1.1.1 and I got the error when compiling the file `arm_nn_mat_mult_s8_nt_t_s8.c` that is using the macro `__SXTB16_RORn`

JonatanAntoni, 2020-06-22:
@felix-johnny can you comment whether CMSIS-NN should compile using Arm Embedded GCC 9.2.1?
This macro causing issues in this case has been added on behalf of CMSIS-NN.

felix-johnny, 2020-06-23:
@JonatanAntoni  I haven't tried that exact version, but it should compile. And it is just part of arm_nn_mat_mult_s8_nt_t_s8.c added by https://github.com/ARM-software/CMSIS_5/pull/796 . 

What is the 'impossible constraint' in the error message? could it be register allocation?

GuentherMartin, 2020-06-23:
Hello @felix-johnny , the only 'impossible constraint' I can think of would be "i" which stands for immediate integer operand.
GCC should support this.

felix-johnny, 2020-09-28:
Related to https://github.com/ARM-software/CMSIS_5/issues/996 where a solution is provided.

closed 2021-01-07 by felix-johnny