ARM-software/CMSIS_5 #884: Queries regarding implementation of cmsis nn

bbhusin, 2020-04-03:
Hii, 
I am trying to use   CMSIS NN  (arm_convolve_s8)  where I have a queries regarding  input parameters out_activation_min, out_activation_max, out_offset, input_offset, buffer_a, output_mult,output_shift.
If I could get any reference example it would be helpful and also any documentation explaining the  functionalities.

felix-johnny, 2020-04-06:
@bbhusin Have you looked at the arm_nnfunctions.h file? It has the ranges for parameters. As for reference example,there are two places that you can check. 

1. CMSIS-NN unit test case for arm_convolve_s8() in https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Tests/UnitTest/TestCases/test_arm_convolve_s8/test_arm_convolve_s8.c 

2. TensorFlow Lite Micro's unit test cases from  https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/kernels/conv_test.cc

felix-johnny, 2020-06-10:
I am closing this issue. Feel free to open one if there are any further questions.

closed 2020-06-10 by felix-johnny