ARM-software/CMSIS_5 #886: Possible overflow in arm_rms_q31()?

dsabdallah, 2020-04-06:
It looks like arm_rms_q31() can quickly overflow its accumulator because it's accumulating a potentially large number of Q2.62 values (that result from squaring Q1.31 values) into a uint64_t accumulator. 

In the code below, should the result of the multiply be scaled before being accumulated as done in the arm_power_q31() function? 

From arm_power_q31():

`sum += ((q63_t) in * in) >> 14U;`

From arm_rms_q31():

`sum += ((q63_t) in * in);`

arm_rms_q15() uses a q63_t accumulator which is large enough to hold many intermediate multiplies but arm_rms_q31() uses uint64_t which doesn't seem like a large enough to hold more than a few Q1.30 multiplies.

Also, are the arm_power_q31() and similar power functions misnamed? They appear to be calculating the energy of the input. If they were calculating power, I would expect them to divide by the block size as done in arm_rms_q31() and similar functions.

Thanks

christophe0606, 2020-04-07:
The documentation for the function is saying that the input should be scaled to avoid saturation:

> In order to avoid overflows completely, the input signal must be scaled down by log2(blockSize) bits

I have not checked if this value is the correct one. But what is sure is that some scaling is required.

It is right that the naming of "power" function is wrong. It is energy. But we can't change the name because it would change the API and break existing code using the library.

dsabdallah, 2020-04-07:
Thanks for the response and my apologies: I recall now reading that line and promptly forgetting it, though it doesn't seem like very strong enough language for a function that can overflow so easily (perhaps just removing the word "completely" would clear that up). However, why does this function not do the scaling but arm_power_q31(), which is the first step in calculating RMS, does scale? This seems oddly inconsistent which could make mistakes like mine more common.

Also, can the documentation for the arm_power_() family of functions be updated to indicate that it is in fact calculating energy? That would at least clear up any confusion others might have regarding its name and behavior.

Thanks

christophe0606, 2020-04-07:
There are lot of inconsistencies and documentation work to do. We are improving the library everyday but it takes time.

I am taking  note for the documentation improvement and creating another github issue for this.

closed 2020-04-07 by christophe0606

dsabdallah, 2020-04-07:
Thanks again. I understand these things take time and appreciate all the effort put into improving the library. Just trying to do my part, as a user, by pointing out potential issues as I find them when it doesn't appear from the issue list that they've been mentioned.