ARM-software/CMSIS_5 #890: Incorrect title in some CMSIS-DSP file headers

stephanosio, 2020-04-13:
Nits: Found some incorrect "titles" in the source file headers

https://github.com/ARM-software/CMSIS_5/blob/a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_f32.c#L3

https://github.com/ARM-software/CMSIS_5/blob/a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_f64.c#L3

https://github.com/ARM-software/CMSIS_5/blob/a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f32.c#L3

https://github.com/ARM-software/CMSIS_5/blob/a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c#L3

christophe0606, 2020-04-14:
Thank you. I'll correct this.

christophe0606, 2020-04-14:
Corrected in latest commits.

closed 2020-04-14 by christophe0606