ARM-software/CMSIS_5 #891: arm_cfft_init_f64.c not listed in CMakeLists.txt

stephanosio, 2020-04-13:
I have noticed that the `arm_cfft_init_f64.c` is not listed in the `CMakeLists.txt`. Is this intentional?

Also the `arm_cfft_init_f64` function is not covered by the test suite -- the current `TransformCF64.cpp` implementation relies solely on the static tables. Is this also intentional?

https://github.com/ARM-software/CMSIS_5/blob/93b52c8a563140ba236b439c3466c4044c404641/CMSIS/DSP/Testing/Source/Tests/TransformCF64.cpp#L44

christophe0606, 2020-04-14:
The new init functions are mandatory for the vectorized version. Since there is no vectorized version for the f64 fft, the old way of initializing is enough.

But, you're right that the new init function should be included in the CMakeLists.txt

And for coherency with other tests, it may be better to use the init function too rather than the structs.

I'll fix this.

christophe0606, 2020-04-14:
Corrected in latest commits.

closed 2020-04-14 by christophe0606