ARM-software/CMSIS_5 #893: DSP: Missing "const"?

TTornblom, 2020-04-15:
A colleague noticed that https://github.com/ARM-software/CMSIS_5/blob/2e98b24711e8d73d425484334f3d54c41168ffd9/CMSIS/DSP/DSP_Lib_TestSuite/RefLibs/src/FilteringFunctions/biquad.c#L161 appears to miss a "const".

All other functions in the file have a "const" qualifier for this line.

Is this by design or just an omission?

christophe0606, 2020-04-15:
I think it was just an omission. Because both instance structure (f32 and f64) are using a const for the coefs.

But we are no more maintaining those C references. We have decided to validate the library by using references generated from scipy or matlab.

TTornblom, 2020-04-16:
Looks like not having a "const" there causes compile errors with the IAR toolchain.

We are still using these tests so I've fixed this in PR #895

christophe0606, 2020-04-16:
Ok. So I'll merge the PR.

closed 2020-04-16 by christophe0606