ARM-software/CMSIS_5 #896: negative value is not shifted correclty

rabbitsaviola, 2020-04-16:
As NN_ROUND is defined as `1u<<shift_value`, it's an unsigned type. It causes product of src and dst (see code below) is logical shifted rather than arithmatic shift.

https://github.com/ARM-software/CMSIS_5/blob/d82d3cc52a2ccccafa55fbc84a6fbc59e5f85531/CMSIS/NN/Source/NNSupportFunctions/arm_nn_mult_q7.c#L79

I use arm compiler v6.10.1, cortex-m33

felix-johnny, 2020-04-28:
@rabbitsaviola Thanks for the question. The range of shift expected is 0 to 31.

arvoelke, 2020-04-29:
I don't think @rabbitsaviola is asking about the range of the shift. If I understand right, they are asking about the case where `(*pSrcA++) * (*pSrcB++)` is negative. The issue being raised is if that if the value being shifted is unsigned (since `NN_ROUND` produces an unsigned value) then wouldn't `>> out_shift` be a logical shift rather than arithmetic shift? If so, then negative products would not be shifted correctly.

arvoelke, 2020-04-29:
Here is a minimal example to help explain what's happening independently of CMSIS:

```C
#include <stdio.h>

int main() {
  int x = -32;
  printf("%d\n", (x >> 1));  // arithmetic shift
  printf("%d\n", (((int) x + 0u) >> 1));  // logical shift (becomes unsigned before shift)
  printf("%d\n", ((int) (x + 0u) >> 1));  // arithmetic shift (cast back to signed before shift)
  return 0;
}
```

Using `gcc` (version 7.5.0) this outputs:
```
-16
2147483632
-16
```

The middle line is not the behaviour we want. This line maps onto above code as the addition of `NN_ROUND(out_shift)` which is unsigned, before doing the shift.

Would you welcome a PR for fixing this issue?

felix-johnny, 2020-04-29:
@arvoelke Thanks for the clarification. Yes, that integer conversion issue needs a fix. You are welcome to create a PR for the fix.

PR #908 (2020-05-04): CMSIS-NN: Use arithmetic shifting in arm_nn_mult
Fixes #896.

I have not tested this directly. Let me know if a unit test is needed (I see some examples/instructions in the [UnitTest](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN/Tests/UnitTest) directory that I could follow). I've tested it indirectly through an example in the issue:

```C
#include <stdio.h>

int main() {
  int x = -32;
  printf("%d\n", (x >> 1));  // arithmetic shift
  printf("%d\n", (((int) x + 0u) >> 1));  // logical shift (becomes unsigned before shift)
  printf("%d\n", ((int) (x + 0u) >> 1));  // arithmet [... cut]
commit e409b450 (2020-05-04): CMSIS-NN: Use arithmetic shifting in arm_nn_mult

Fixes #896.
closed 2020-05-04 by felix-johnny