ARM-software/CMSIS_5 #900: PackChk.exe is outdated

Vincent-vanDiddenDC, 2020-04-21:
The PACK.xsd is updated with the latest addition of the Cortex-M55.
However, when trying to install as a pack in eclipse the following errors arrise on multiple lines:
- "cvc-enumeration-valid: Value 'Cortex-M55'is not facet-valid with respect to enumeration ..."
- "cvc-attribute.3: The value ‘Cortex-M55’ of attribute ‘Dcore’ on element ‘processor is not valid with respect to its type, 'DcoreEnum’."
In the error the Cortex-M55 is missing from the enumeration, suggesting that the Pack Checker is still outdated.

edriouk, 2020-04-21:
Hi Vincent, 

The CMSIS Eclipse plug-ins neither contain nor use PackChk.exe. 
The Coretex-M55 is not supported in 2.5.1, but planned for 2.6.0 (coming soon)
However the support is only limited by Arm Compiler Toolchain adapter, not other parts of the plug-in.
What pack installer do you use?  
Is the updated PACK.xsd is used for schema check? 
Best regards,
Evgueni

JonatanAntoni, 2020-04-21:
Hi @Vincent-vanDiddenDC,

which version of Eclipse are you using? The messages you are seeing are not related to PackChk.exe. 

The CMSIS Plugins for Eclipse (official release) have not yet been updated to support latest enhancements to PACK.xsd. Please refer to https://github.com/ARM-software/cmsis-pack-eclipse. 

Cheers,
Jonatan