ARM-software/CMSIS_5 #903: Cannot clone this repository

MateuszKlatecki, 2020-04-22:
Hi,

Since yesterday I cannot clone this repository. Neither develop branch nor tags. 
It looks like a problem with git-lfs.

Here are the logs:
```sh
$ git clone https://github.com/ARM-software/CMSIS_5.git -b develop
Cloning into 'CMSIS_5'...
remote: Enumerating objects: 113, done.
remote: Counting objects: 100% (113/113), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 66937 (delta 44), reused 37 (delta 12), pack-reused 66824
Receiving objects: 100% (66937/66937), 228.29 MiB | 9.65 MiB/s, done.
Resolving deltas: 100% (50020/50020), done.
Updating files: 100% (4782/4782), done.
Downloading CMSIS/Utilities/Win32/PackChk.exe (2.9 MB)
Error downloading object: CMSIS/Utilities/Win32/PackChk.exe (7934974): Smudge error: Error downloading CMSIS/Utilities/Win32/PackChk.exe (7934974e4e929021a7413a609183ce61142c38d9c87e40925a0feec1bc958892): [7934974e4e929021a7413a609183ce61142c38d9c87e40925a0feec1bc958892] Object does not exist on the server: [410] Object does not exist on the server

Errors logged to C:\Users\klata\Desktop\CMSIS_5\.git\lfs\logs\20200422T075138.7392037.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: CMSIS/Utilities/Win32/PackChk.exe: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

JonatanAntoni, 2020-04-22:
Hi @MateuszKlatecki,

thanks for reporting back. This is an issue caused by moving away from Git-LFS.
Can you please try to disable Git-LFS smudge filter?

Cheers,
Jonatan

MateuszKlatecki, 2020-04-22:
@JonatanAntoni thank you for your help, Solution you provided helped me

JonatanAntoni, 2020-04-22:
Sorry for any inconvenience resulting from this fundamental change. Please leave this issue open until we finally solved it. I am still looking for a solid solution. The current situation might affect all users referring to this repository. 

I am happy for any idea how we could distribute binary utilities (such as PackChk and SVDConv) in the future.

malferns, 2020-04-22:
Are we not expected to run the git lfs commands any more after cloning the CMSIS_5 git repository?
'git lfs install --local && git lfs pull'

JonatanAntoni, 2020-04-23:
We removed all files formerly stored in Git-LFS due to bandwidth quota issues. So no need to run lfs commands any more.

GaborAbonyi, 2020-04-23:
Is there any other way to get the binaries of previous releases? The release tarballs does not contain the binaries.

JonatanAntoni, 2020-04-23:
Yes, all binaries are in the released pack, see https://github.com/ARM-software/CMSIS_5/releases/download/5.7.0/ARM.CMSIS.5.7.0.pack.

Unfortunately you are right, the source tarball only contains the pointer-files. Too bad.

GaborAbonyi, 2020-04-23:
Thanks! I only checked the tar.gz and zip files before, but the pack archive seems to contain all the necessary binaries.

malferns, 2020-04-25:
Is this applicable to all branches/tags?  We have been using the 5.5.0 tag for trusted-firmware builds.

Previously used commands:
git clone https://github.com/ARM-software/CMSIS_5.git -b 5.5.0 && cd CMSIS_5 && git lfs install --local && git lfs pull && cd -

What commands should we use now?

malferns, 2020-04-25:
This is the error I get now:

+ git clone https://github.com/ARM-software/CMSIS_5.git -b 5.5.0
Cloning into 'CMSIS_5'...
Note: checking out 'a82c293bfb9f07285dcdf93507e5559acc0f92af'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Downloading CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib (2.4 MB)
Error downloading object: CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib (afa93f0): Smudge error: Error downloading CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib (afa93f014c4252cf0c530b223264c5d0b9b5d06b1fd02cce0e68976ed2877b5e): [afa93f014c4252cf0c530b223264c5d0b9b5d06b1fd02cce0e68976ed2877b5e] Object does not exist on the server: [410] Object does not exist on the server

Errors logged to /var/tmp/gitlab-runner1/builds/7e5c9300/1/repo/tfm/trusted-firmware-m/CMSIS_5/.git/lfs/objects/logs/20200424T204818.428715277.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: CMSIS/DSP/Lib/ARM/arm_ARMv8MBLl_math.lib: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.

JonatanAntoni, 2020-04-27:
Hi @malferns,

yes, all branches/tags are affected, currently. I am going to fix this, today.

Sorry for any inconvenience,
Jonatan

JonatanAntoni, 2020-04-27:
Hi all,

we have decided to bring back all files to LFS storage to assure backward compatibility for the time being. Checking out older branches/tags should be possible, again.

In order to reduce bandwidth consumption I want to ask you to reduce cloning from GitHub to a bare minimum. Please consider adding caches to your CI systems for instance, i.e. keep a bare clone somewhere local.

Cheers,
Jonatan

malferns, 2020-04-28:
I've verified that my previous commands work once again with the upstream GitHub repo.

We do maintain a bare mirrored repo which is used for our CI builds, but I had temporarily switched to the upstream repo as our mirror was broken because of these changes.  

The mirror has been fixed now and we are back to using the mirror for our CI builds.

Thanks for reverting the changes.

JonatanAntoni, 2020-04-28:
Hi @malferns,

thank you for your confirmation.

Cheers,
Jonatan