ARM-software/CMSIS_5 #906: RIFFT source buffer access mishap

stephanosio, 2020-04-29:
When the `arm_rfft_*` functions execute inverse FFT, the source buffer is accessed beyond what would be considered an acceptable limit (i.e. input data length).

### Overview (using `arm_rfft_q31` as an example)

1. `arm_rfft_q31` calls `arm_split_rifft_q31` with `fftLen = fftLenReal / 2`.
https://github.com/ARM-software/CMSIS_5/blob/c5ff98f5254a00a75652abbdc1d2dd650725abdc/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q31.c#L86-L93

2. `arm_split_rifft_q31` sets `pIn2` to `2 * fftLen + 1`, which is equal to `fftLenReal + 1`.
https://github.com/ARM-software/CMSIS_5/blob/c5ff98f5254a00a75652abbdc1d2dd650725abdc/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q31.c#L439

3. `arm_split_rifft_q31` reads `pIn2`, which is effectively two elements beyond the end of the input data.
https://github.com/ARM-software/CMSIS_5/blob/c5ff98f5254a00a75652abbdc1d2dd650725abdc/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q31.c#L474

### Impact

This effectively requires the input/source buffer length to be at least `fftLenReal + 2` and the two extra elements to be initialised to a safe value (i.e. 0).

For instance, if immutable input data is simply copied to a mutable source buffer, which is sized identical to the input data, the behaviour of this function is unpredictable because there can be any data beyond the end of the source buffer.

### Suggested solution

1. In the documentations, clarify that the source buffer length must be at least `fftLenReal + 2` and the last two elements must be initialised to 0.
2. Fix the RIFFT implementation, so that no such maneuvre is required.

christophe0606, 2020-05-04:
Thank you for the detailed report. As soon as I can find some time (hopefully soon), I'll correct it.

christophe0606, 2020-05-29:
I have updated the comments in commit https://github.com/ARM-software/CMSIS_5/commit/217eddd032753519e62ea12ccf2f25caae0ea8cc

But according to what is computed by the arm_split_rfft_(f32|q31|q15), the last two elements should not be 0.

For instance, for arm_split_rfft_q31, the values are:

```
pDst[2U * fftLen] = (pSrc[0] - pSrc[1]) >> 1;
pDst[(2U * fftLen) + 1U] = 0;
```

So, I have written this in the comments.

stephanosio, 2020-07-02:
Closing since the buffer access rules have been clarified in the documentation.

closed 2020-07-02 by stephanosio