ARM-software/CMSIS_5 #910: arm_sorting.h: No such file or directory

AnXelion, 2020-05-06:
Hello Arm, 

I'm trying to do [this tutorial: Image recognition on Arm Cortex-M with CMSIS-NN
](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/single-page) provided by Arm.

During the compilation, the following error messages show up. It seems like it's related to some sorting stuff which is added in the release v5.7.0.

```
Compile [ 75.5%]: arm_bubble_sort_f32.c
[Fatal Error] arm_bubble_sort_f32.c@30,25: arm_sorting.h: No such file or directory
[ERROR] ../CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_bubble_sort_f32.c:30:25: fatal error: arm_sorting.h: No such file or directory
 #include "arm_sorting.h"
                         ^
compilation terminated.
```

Any suggestions to solve this issue? Thanks!

christophe0606, 2020-05-06:
You need to include the folder DSP/PrivateInclude when building.

This folder is containing headers which are used only for building the libraries but not to use it.

The other folder DSP/Include is for headers which are needed to use the library.

I don't know how the mbed build is working. The build was tested with the latest MDK and with the DSP cmake files.

AnXelion, 2020-05-12:
After including the folder DSP/PrivateInclude when building, it works. 
Thanks for the help. Really appreciate it.

closed 2020-05-12 by AnXelion

OrangeForTangerine, 2021-07-22:
感谢提醒

QM1412, 2023-07-25:
where can i find it ?

christophe0606, 2023-07-25:
@QM1412 CMSIS-DSP now has its own github repository. And the file is here : https://github.com/ARM-software/CMSIS-DSP/tree/main/PrivateInclude

QM1412, 2023-07-25:
ohoh，find it ,thanks