ARM-software/CMSIS_5 #911: DSP missing vector files

GI101, 2020-05-06:
Hello,
After compiling DSP package the ControlerFucntions folder vector files were not included in the build. The Clarke/Park vector files are shown in the drop down list and should be part of the DSP package download zip file. It would seem intentional and ARM will be notified these files were omitted from the project build if they are not added as required. 

[image]
[image]

christophe0606, 2020-05-07:
Those functions are inline functions and they are defined in arm_math.h.
So they are included in the package and the source code is in arm_math.h

If you want to use them, you'll need to include arm_math.h. The functions will not be in the binary library.

GI101, 2020-05-07:
Hi Christopher,

Thanks for the fast reply!

Oddly I reviewed arm_math.h and did not off hand notice clarke/park inline functions or would not have posted. The in lines are controller functions perhaps should have linked arm_math.h into the folder path noted. TI had not updated CMSIS since 2015 and had the older math.h (2012) which I replaced with arm_math.h (2019) prior to building the project. Technically all H files are part of the package especially inline functions. So the user has to hunt the includes folder for the H file just to review in IDE editor to review the syntax structure and method. 

Regards,

GI101, 2020-05-07:
BTW: Clarke transform seemingly does not represent Kirchhoff's current law; for  Ia+Ib = Ic or any other combination of 3 inputs, never = 0 but only when phases are switched off or 50% PWM duty cycle. Where do we input Ic for inline arm_park_f32()? It would seem inline park call is outdated,  being more typical to input ADC samples from 3 channels with modern FOC controllers.

ARM Note:
Stator current space vector and its components in (a,b). and Ia + Ib + Ic = 0, in this condition Ialpha and Ibeta can be calculated using only Ia and Ib.  

  __STATIC_FORCEINLINE void arm_park_f32(
  float32_t Ialpha,
  float32_t Ibeta,
  float32_t * pId,
  float32_t * pIq,
  float32_t sinVal,
  float32_t cosVal)
  {
    /* Calculate pId using the equation, pId = Ialpha * cosVal + Ibeta * sinVal */
    *pId = Ialpha * cosVal + Ibeta * sinVal;

    /* Calculate pIq using the equation, pIq = - Ialpha * sinVal + Ibeta * cosVal */
    *pIq = -Ialpha * sinVal + Ibeta * cosVal;
  }

GI101, 2020-05-08:
One would think 3 phase infers not 2 but 3 current phases. The reason for adding the 3rd phase is to achieve 100% duty cycle without having to handle issue of current reconstruction. I modified the DSP Park transform to what seems an logical approach to handle (Ic) input.  Well it needs to be tested but should allow 3 phases to control SVPWM sector rotation angles. 

  /**
   *
   * @brief  Floating-point Clarke transform
   * @param[in]  Ia       input three-phase coordinate <code>a</code>
   * @param[in]  Ib       input three-phase coordinate <code>b</code>
   * @param[in]  Ic       input three-phase coordinate <code>c</code>
   * @param[out] pIalpha  points to output two-phase orthogonal vector axis alpha
   * @param[out] pIbeta   points to output two-phase orthogonal vector axis beta
   * @return        none
   */
  __STATIC_FORCEINLINE void arm_clarke_f32(
  float32_t Ia,
  float32_t Ib,
  float32_t Ic,
  float32_t * pIalpha,
  float32_t * pIbeta)
  {
    /* Calculate pIalpha using the equation, pIalpha = Ia */
    *pIalpha = Ia;

    /* Calculate pIbeta using the equation,
     * 	pIbeta = (1/sqrt(3)) * Ia + (2/sqrt(3)) * Ib + (3/sqrt(3)) * Ic  */
    *pIbeta = (((((float32_t) 0.57735026919) * Ia) +
    		     ((float32_t) 1.15470053838) * Ib) +
			     ((float32_t) 1.73205080757) * Ic);
  }

christophe0606, 2020-05-11:
Thank for the feedback.

The clarke and park transforms have not been updated for a very long time.
And it is not at all my domain so I can't comment.

But if there is the need to improve the implementation then we are totally ok to do it when we can find some time (so unfortunately it may not be quick since we have lot of other things to do on the library currently).

So, I move this ticket to enhancement to track it.

GI101, 2020-05-13:
Yes it seems a bit different from many that are now embedded with M7 Cortex. Wonder that being a vendor or ARM factory load to ROM? For instance NXP uses below model and it seems was coded from arm_math.h inline calls. The bigger problem being package pinouts on many M7 packages don't match the one of bare PCB we have to update from M4 class. So we have to code it to test it with our DC inverter, not the ones a vendor tires to sell for evaluation only. That is like comparing a Kuala with a polar bear and suspect can or may end badly.  

Thanks Christophe!

[image]

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606