ARM-software/CMSIS_5 #920: BIAS can be NULL in Tensorflow Micro, however this case is not handled by CMSIS

oleksii-abramenko, 2020-05-15:
Hello everyone,

In Tensorflow Micro the function  arm_fully_connected_s8(...) can be called with argument bias==nullptr. However, this case is not handled by CMSIS. If we look at implementation of arm_fully_connected_s8() we can note that it uses arm_nn_vec_mat_mult_t_s8().

Inside arm_nn_vec_mat_mult_t_s8() there are no checks enforced on value of bias, whether it is nullptr or not.  Consequently, it may happen that one may be reading from nullptr location(as well as iterating over neighboring memory cells), and using this "data" in the computations. 

So, if in your computational graph the bias is absent at any place, the whole inference will be messed up.

felix-johnny, 2020-05-25:
@oleksii-abramenko Thanks for raising this issue. You are right and this needs to be fixed.

felix-johnny, 2020-06-10:
To be correct on TFLu for now , I have made this PR(https://github.com/tensorflow/tensorflow/pull/40269) to not use the CMSIS-NN API when bias is not null.

felix-johnny, 2020-09-28:
Thanks again for raising the issue. It is now resolved.

closed 2020-09-28 by felix-johnny