ARM-software/CMSIS_5 #921: Autocorrelation in DSP Filtering Functions

alvarofrutos, 2020-05-15:
It would be nice to have an [autocorrelation](https://en.wikipedia.org/wiki/Autocorrelation) function in the DSP Filtering Functions.

This function would accept just one input sequence and its length. The output array could be reduced to the half due to it being symmetrical.

This algorithm would just need to calculate **half of the output data**, improving its efficiency.The checks on the lengths of the sequence could be also removed.

```
/**
  @brief         Autocorrelation of floating-point sequences.
  @param[in]     pSrc       points to the input sequence
  @param[in]     srcLen     length of the input sequence
  @param[out]    pDst       points to the location where the output result is written.  Length srcLen.
  @return        none
 */
void arm_autocorrelate_f32(
  const float32_t * pSrc,
        uint32_t srcLen,
        float32_t * pDst);
```

christophe0606, 2020-05-28:
Good suggestion. I have added it to our list of enhancements for the library (but we have no planning we can communicate).

christophe0606, 2022-06-30:
We are moving CMSIS-DSP to a new repository : https://github.com/ARM-software/CMSIS-DSP

I have recreated the github issues on the new repository (sometimes merging a few related ones). You can see the link just above this comment (... mentioned this issue ...).

So, I am closing the github issue here.

closed 2022-06-30 by christophe0606