ARM-software/CMSIS_5 #927: Is there a plan to implement dilation in convolution/depthwise conv?

majianjia, 2020-05-23:
Hi, 
I am the author of [NNoM](https://github.com/majianjia/nnom) which runs on top of CMSIS-NN and taking advantage of its fast speed.

Recently many developers have asked me whether dilated conv is supported. Dilated convs have many advantages in voice recognition for example.  I see CMSIS-NN has already passing dilation_x/dilation_y arguments in many conv files but they are ignored in the codes.    https://github.com/ARM-software/CMSIS_5/blob/1150e71e07c79b538efd842aba5b210a31827ae5/CMSIS/NN/Source/ConvolutionFunctions/arm_depthwise_conv_s8.c#L240

Since we are moving to a new interface in `*_s` (A big thumb up to the new API BTW). Is there a plan to implement dilatated conv soon?

We are really like to see/use the CMSIS version for faster interference than the current pure C version I have in NNoM.  
Thanks, 
Jianjia

felix-johnny, 2020-05-25:
@majianjia Thanks for the question and the thumbs up on the new API! 
  Adding support for dilated conv is not in the immediate plans, but if you have models in the embedded space that use dilation, let me know and we will consider it as a future enhancement. In most functions(arm_convolve_s8 for e.g), the performance penalty should be minor.

majianjia, 2020-05-25:
@felix-johnny 
Thank you for your quick reply. 
I started to looking for this because many developers contact me are using NN for voice recognition in their products. For example, WaveNet and TDNN (with conv) are using 1-D dilated Conv for MFCC. Many prebuilt nets for image segmentation are also using dilation on CNN. 
So I think there are some need for it. 
Its very good you could confirm the penalty will not be much. I will have a look in detail when I have time and see if I could help. 
Thanks