ARM-software/CMSIS_5 #934: Why use MEAN_DATA in the cmsis_nn model

ilia19911, 2020-05-27:
I tried to convert the CAFFE model using the provided nn_quantizer.py and code_gen.py scripts, preliminarily adapting them to python3 (in the root of the folder by reference). Next, I assembled a project under GCC for linux (folder cmsis_nn_on_pc). And each time with a new quantization, I get completely different accuracy (from 50% to 90%, At the same time, caffe accuracy is about 98%
). However, I do not understand 
1) why is it necessary to use MEAN_DATA in the cmsis_nn model if these are average values ​​for the training image database? 
2) Why do you need to use DATA_RSHIFT if this also applies only to training values, and for new ones, you need to subtract the average and normalize each color of each image separately?
3) Why is DATA_RSHIFT often a NEGATIVE NUMBER ??
4) Why is there no image preprocessing in your example?

And the softmax function does not work without __usat asm instructions, although somewhere there is a reader that cmsis nn is easily assembled for the PC when using MATH_CM0. Therefore, I just had to compare the output values ​​of the neural network

https://1drv.ms/u/s!AipNn2xvZES_k6MDIjPaq-Nw48sG5A?e=PL6az4

ilia19911, 2020-06-18:
Thank. Useful article. I was able to fix all my mistakes. In jpeg. then the data really helps to improve accuracy. At the moment, my network indicates the presence of clouds in the photo with an accuracy of 97%. But for some reason, we had to replace the string in the Run_nn function 
    image_data[i] = (q7_t)__SSAT( ((int)(image_data[i] - mean[i]) << 1), 8); 
 with    
    image_data[i] = (image_data[i] - mean[i]) ;

I understand that this is bad, since there is no check for oversaturation, but the accuracy is good. I did not go deep yet, since without this the result is almost like in the original model. It is possible that uint32_t is returned in __SSAT, although the uint8_t array is used for the image array. Can you tell me something?

felix-johnny, 2020-09-28:
@ilia19911 Sorry about the delay. Glad you were able to get it to work! About SSAT, It returns an int<x> as result. 
https://www.keil.com/support/man/docs/armasm/armasm_dom1361289904320.htm

 I would check the data types to see any undesired data conversions is going on there.

ilia19911, 2020-09-28:
@felix-johnny  
I did not see any problems with the conversion. SSAT works correctly over the entire range. But could there be a problem with the caffe settings? can the network train there without checking for saturation, and just an overflow occurs?

felix-johnny, 2021-01-07:
@ilia19911  Sorry, I do not have an answer for that. I am closing this ticket and if you have any questions please let us know!

closed 2021-01-07 by felix-johnny